wokwi/wokwi-features #123: Library request

SteveRMann, 2021-06-24:
Glad I found this simulator.
Any chance of adding the ESP8266 and ESP8266WiFi libraries?

urish, 2021-06-24:
Thank you for the comment!

You mean full ESP8266 simulation?

SteveRMann, 2021-06-24:
Almost everything I do with the Android IDE is on ESP devices (Wemos D1, D1 mini, NodeMCU and various Sonoff devices).
I only use the Uno to test code segments.
I can appreciate that it would be a challenge to simulate WiFi, but maybe a terminal-like interface?  At least enough to compile without error.

A downside I've realized is that the simulator doesn't tell me what or where my coding errors are.

urish, 2021-06-24:
> I can appreciate that it would be a challenge to simulate WiFi, but maybe a terminal-like interface? At least enough to compile without error.

It is, indeed - but there's an initial, work-in-progress implementation for the [Nano RP2040 Connect chip].

The challenge is that ESP8266 uses a totally different architecture (called Xtensa), and implementing a new architecture is literally months of work. I'm now working on the RP2040 (for the new Pi Pico), and it's been going on since January. If you are interested, there's a [playlist](https://www.youtube.com/playlist?list=PLLomdjsHtJTxT-vdJHwa3z62dFXZnzYBm) where you can see how the RP2040 simulation implemented.

> A downside I've realized is that the simulator doesn't tell me what or where my coding errors are.

It should tell you - so if it doesn't this is definitely a bug. Can you share an example where it doesn't tell you about an error?

sutaburosu, 2021-06-24:
I certainly see errors, but no warnings are emitted.  arduino-cli [defaults to no warnings](https://arduino.github.io/arduino-cli/latest/commands/arduino-cli_compile/).

lemakavely, 2022-01-10:
soo we cant add the library for ESP8266WiFi?

urish, 2022-01-10:
You can add the library itself, but we don't have an ESP8266 part, so it probably won't be very useful...

urish, 2022-01-27:
Closing as we have ESP32 with WiFi. 

There's also an open request for ESP-01 WiFi AT MODEM in #169

closed 2022-01-27 by urish