wokwi/wokwi-features #124: The pins wont activate (for routing) if I change number of LEDs in the Ring

matititam, 2021-06-24:
**Project link**: https://wokwi.com/arduino/projects/302135681537278473

1. Add a LED Ring in the simulator window
2. Make sure the pins are active for routing using the mouse
3. Change the number of LEDs to other than 16 (example: 12)
4. check the pins now. the active connections won't appear
5. below image for reference

[image]

urish, 2021-06-24:
Thanks for reporting!

closed 2021-06-24 by urish

drf5n, 2022-04-16:
Since wokwi-led-ring was mentioned here, I thought I'd paste the json used in that sketch:

```
    {
      "type": "wokwi-led-ring",
      "id": "ring1",
      "top": 11.33,
      "left": -182,
      "rotate": 0,
      "hide": false,
      "attrs": { "pixels": "16" }
    }
```

It woks fine for "pixels": "160"