wokwi/wokwi-features #125: Lock project improvements

arcostasi, 2021-06-24:
When activating the Lock Project option, it would be interesting to lock the diagram in the same way as it is done when running the simulation. Another suggestion is to hide the "Add a new part" button. This way it is possible to embed simulations in articles and  tutorials online with parameter "?view=diagram" without the user deleting something by mistake.

urish, 2021-06-24:
Should we make this part of the Lock option? Because then it means that if we shared a locked project, people won't be able to tinker with it before creating their own copy. 

Another option would be to have a special "view-only" mode where you don't have the ability to edit the code / diagram, and a "open in edit mode" button would switch to edit mode

arcostasi, 2021-06-24:
Would the "Block Project" option be the simplest to implement? I think I don't need to lock the code editor and JSON, just hide the button and lock the diagram so as not to erase the wires when clicking or moving parts.

urish, 2021-06-24:
What do you refer to when you say "Block project"?

Your use case is embedding the simulation, so it may need to behave differently than when generally sharing a link. 

What would be, for you, the optimal behavior for the embedded widget?

arcostasi, 2021-06-25:
> What do you refer to when you say "Block project"?
Sorry Uri, "Lock project" which is located in the Save menu.

> Your use case is embedding the simulation, so it may need to behave differently than when generally sharing a link.
I had thought of something simpler, just use the features that are already available in the simulator to lock the diagram together with the parameter "?view=diagram" implemented.

> What would be, for you, the optimal behavior for the embedded widget?
Just display the diagram and simulate the operation.
the code is something easy to implement on the site and be able to highlight parts.

urish, 2021-06-27:
> Just display the diagram and simulate the operation.
> the code is something easy to implement on the site and be able to highlight parts.

I'm asking because, as I see things, the big advantage of Wokwi is that it lets users play around with code and explore things themselves. 

I'm trying to think - if we only display a read-only version of the diagram, what advantage does it have over let's say embedding a YouTube video (or even a GIF) that shows you the result?

arcostasi, 2021-06-28:
I understand your point of view Uri! maybe it's not a good choice to lock the diagram. this reminds me of the phrase of the great teams "Team that is winning, don't move." :)

closed 2023-09-29 by urish