wokwi/wokwi-features #128: Paypal button on login screen.

leftCoast, 2021-06-30:
Not happy about signing up for pay/week thing. But.. A PayPal button that is somewhere, easy to see? Doesn't have to be on the login screen. And.. And now that I'm thinking about it. A priority list of features with PayPal buttons on them! So we can "Vote with our dollars" for the next feature. Sorted by amount of cash. Maybe you could collect 50% as a deposit then get the remaining 50% when added?

I'll pony up the Adafruit 1947 with WORKING SD drive for $100. Maybe others will add to this? Kinda' like crowd funding.

Actually, I'd pony up for the SD Drive alone for $100. Then it would be easier to add to the 1947 seeing it's already completed.

urish, 2021-07-03:
That's an interesting idea!

One thing I'm worried about is that it'll add the complexity of managing the deposits, refunds, etc.

Perhaps we can come up with a simpler model? Any ideas?

Koepel, 2021-07-31:
Voting for a certain feature with dollars has a downside. Suppose someone puts down a lot of money for a Nextion display simulation. That could stop the normal Wokwi development for weeks or months.

Could I, for example, give money to undo the $100 for the Adafruit TFT display simulation ?

leftCoast, 2021-08-02:
The problem with US elections it that, although there were many people running, the general consensus is that you vote for A (Terrible) or B(Horrible) or you've lost your vote. Why? Because no one else could get enough votes to win. My thought, and that of many others, is we need to be able to cast a negative vote to the candidate we hate the most. This way Terrible & Horrible can be reduced enough to at least get an option C. Whatever option C may be.

So in light of this, I say YES. Let people vote dollars againced something you don't want worked on. Although I don't know how that would work. When do you give the dollars? When its not done? That would be now..

urish, 2021-09-17:
> A priority list of features with PayPal buttons on them! So we can "Vote with our dollars" for the next feature.

It took a little while, but now we have it!

https://wokwi.com/features

I added a few sentences at the top of explain how it works. Would love to get your feedback!

urish, 2021-09-17:
> Voting for a certain feature with dollars has a downside. Suppose someone puts down a lot of money for a Nextion display simulation. That could stop the normal Wokwi development for weeks or months.

For now, the process work like this:

1. Someone opens an issue
2. I review the request, and see if it's something that seems feasible / makes sense to other users as well 
3. If all seems good, I add the "open for vote" label on GitHub. Then the feature appearing on the [voting page](https://wokwi.com/features)
4. We work on whatever is at the top of the list. occasional bugs and small changes that take only a couple of minutes may sneak in, but in general the idea is to use the list to prioritize the work.

We'll see how this goes!

leftCoast, 2021-09-17:
I LOVE it! I just added SD cards lets see if it gets any votes.\\

-jim lee

urish, 2021-09-17:
So far, it seems like people really want a way to organize their projects :)

leftCoast, 2021-09-17:
Yeah, I see that. I'm realizing all this organization of files opens up en
entire can of worms. How much does one share? If you want to share a
simulation, you need to share all the bits to build it. And then, if you
are teaching someone, THEY need all the bits as well when they build
something. This can get very involved if you're not careful.

My SD card feature has similar issues. It needs to be an online set of
folders or the simulation can not be shared.

I hope this nets you some sort if real cashflow. One gets so tired of the
internet trying to pay for everything through praise.

-jim lee

urish, 2021-09-17:
Agreed :)

Right now, I'm leaning towards "collections" instead of folders, just added a comments with my thoughts on #100

closed 2021-09-17 by urish