wokwi/wokwi-features #132: PC Fan Simulation

daVinci2793, 2021-07-03:
I would like to respectfully request the addition of a PC Fan simulation. 

To be specific, I would love to see the standard 3-pin and 4-pin 12v 120mm PC fan added. These usually come with a tachometer and PWM control signal. 

Here is a [white paper](https://noctua.at/pub/media/wysiwyg/Noctua_PWM_specifications_white_paper.pdf) from Noctua describing the PWM specification and tachometer functionality. 

I know this is potentially a complicated request, but I love Wokwi and have thus far found it to be easily the best free Arduino Simulator (and for that matter better than all the paid simulators I have tried.) I really appreciate the whole project and plan on using your simulator for teaching and learning for the foreseeable future.

urish, 2021-07-04:
Thanks for the information!

Here are the important bits from the datasheet:

## PWM Input

> Target frequency: 25kHz, acceptable range 21kHz to 28kHz
> Allowed duty-cycle range 0% to 100%
> The signal is not inverted, 100% PWM duty cycle (= 5V DC) results in maximum fan speed.
> The fans operate at full rated speed if there is no PWM input signal.
> 
> The fan‘s speed scales broadly linear with the duty-cycle of the PWM signal between
maximum speed at 100% PWM and the specified minimum speed at 20% PWM (see
www.noctua.at for individual fan specifications): For example, the NF-A12x25 PWM has
a maximum speed of 2000rpm and a minimum PWM speed of 450rpm, so it will run at
2000rpm at 100% PWM, 450rpm at 20% PWM and around 1100rpm at 50% PWM.
> Operation below 20% PWM duty-cycle is not officially supported in the Intel specification
(undefined behaviour). However, most Noctua PWM fans can be operated at below 20%
and will stop at 0% duty-cycle. Only the following models keep running at their specified
minimum speed when the input is below 20%: NF-A20 PWM, NF-S12B redux 1200 PWM
and NF-B9 redux 1600 PWM

## RPM Speed Signal

> Therefore, the formula for obtaining correct RPM speed is: 
> fan speed [rpm] = frequency [Hz] × 60 ÷ 2

daVinci2793, 2021-07-07:
Yeah, that is about all of the useful information I personally gleaned from those documents. Thank you for the response!

urish, 2021-07-07:
Do you have some ideas how to implement the visual part of the PC Fan?

daVinci2793, 2021-07-08:
Well, I have my two cents, but it definitely doesn't need to be anything fancy. Probably just a simple "rotating fan" animation with a direction and speed indicator. Just something that gives a decent visual feedback without being overly complicated. The most important part of this addition would be properly simulating all the inputs/outputs which aren't strictly unique to PC Fans.

yepher, 2022-01-06:
In case it helps I drew this PC Fan in case it can help. To animate just change `id="pc_fan_cvs-fan"` rotate to 0-360 degrees. I can probably do better on spacing of the fan blades if it triggers anyone OCD.

`pc_fan.svg`
```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE svg PUBLIC "-//W3C//DTD SVG 1.1//EN" "http://www.w3.org/Graphics/SVG/1.1/DTD/svg11.dtd">
<svg version="1.1" xmlns="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink" width="288" height="288"  xml:space="preserve" id="pc_fan_cvs">
    <!-- Generated by PaintCode - http://www.paintcodeapp.com -->
    <g id="pc_fan_cvs-fan_page">
        <g id="pc_fan_cvs-frame">
            <rect id="pc_fan_cvs-rectangle2" stroke="rgb(22, 22, 22)" stroke-width="12" stroke-miterlimit="10" fill="rgb(22, 22, 22)" x="0" y="0" width="288" height="288" />
            <circle id="pc_fan_cvs-oval2" stroke="rgb(151, 151, 151)" stroke-width="1" stroke-miterlimit="10" fill="rgb(255, 255, 255)" cx="143.5" cy="144.5" r="137.5" />
        </g>
        <g id="pc_fan_cvs-fan" transform="translate(144, 144) rotate(-124)" >
            <ellipse id="pc_fan_cvs-fan_hub" stroke="rgb(151, 151, 151)" stroke-width="1" stroke-miterlimit="10" fill="rgb(22, 22, 22)" cx="0" cy="0.5" rx="44" ry="40.5" />
            <path id="pc_fan_cvs-fan_blade_01" fill-rule="evenodd" stroke="rgb(18, 18, 18)" stroke-width="1" stroke-miterlimit="4" fill="rgb(18, 18, 18)" d="M -34.94,-108.84 C -25.46,-114.04 -9.43,-116.66 13.13,-116.72 28.32,-116.16 38.18,-112.97 42.7,-107.14 45.33,-105.39 45.33,-95.41 42.7,-77.21 42.3,-69.12 41.03,-61.27 38.9,-53.68 36.11,-45.08 36.11,-45.08 31.59,-37.7 27.73,-37.7 21.97,-43.14 13.13,-45.08 5.31,-47.28 -1.42,-47.58 -4.29,-48.34 -8.04,-55.59 -7.17,-57.85 -11.47,-66.69 -14.89,-73.73 -18.77,-80.54 -23.13,-87.14 L -33.02,-100.36 C -36.44,-104.92 -37.08,-107.74 -34.94,-108.84 Z M -34.94,-108.84" transform=""  />
            <path id="pc_fan_cvs-fan_blade_02" fill-rule="evenodd" stroke="rgb(18, 18, 18)" stroke-width="1" stroke-miterlimit="4" fill="rgb(18, 18, 18)" d="M -41.51,-111.58 C -32.02,-116.78 -16,-119.4 6.56,-119.46 21.75,-118.9 31.61,-115.71 36.14,-109.88 38.77,-108.13 38.77,-98.15 36.14,-79.95 35.74,-71.86 34.47,-64.01 32.33,-56.42 29.54,-47.82 29.54,-47.82 25.02,-40.44 21.17,-40.44 15.4,-45.88 6.56,-47.82 -1.26,-50.02 -7.98,-50.32 -10.85,-51.08 -14.61,-58.33 -13.74,-60.58 -18.03,-69.43 -21.45,-76.47 -25.34,-83.28 -29.7,-89.88 L -39.59,-103.1 C -43,-107.66 -43.64,-110.48 -41.51,-111.58 Z M -41.51,-111.58" transform="rotate(-55.48)"  />
            <path id="pc_fan_cvs-fan_blade_" fill-rule="evenodd" stroke="rgb(18, 18, 18)" stroke-width="1" stroke-miterlimit="4" fill="rgb(18, 18, 18)" d="M -34.94,-108.84 C -25.46,-114.04 -9.43,-116.66 13.13,-116.72 28.32,-116.16 38.18,-112.97 42.7,-107.14 45.33,-105.39 45.33,-95.41 42.7,-77.21 42.3,-69.12 41.03,-61.27 38.9,-53.68 36.11,-45.08 36.11,-45.08 31.59,-37.7 27.73,-37.7 21.97,-43.14 13.13,-45.08 5.31,-47.28 -1.42,-47.58 -4.29,-48.34 -8.04,-55.59 -7.17,-57.85 -11.47,-66.69 -14.89,-73.73 -18.77,-80.54 -23.13,-87.14 L -33.02,-100.36 C -36.44,-104.92 -37.08,-107.74 -34.94,-108.84 Z M -34.94,-108.84" transform="rotate(-122.3)"  />
            <path id="pc_fan_cvs-fan_blade_2" fill-rule="evenodd" stroke="rgb(18, 18, 18)" stroke-width="1" stroke-miterlimit="4" fill="rgb(18, 18, 18)" d="M -34.94,-108.84 C -25.46,-114.04 -9.43,-116.66 13.13,-116.72 28.32,-116.16 38.18,-112.97 42.7,-107.14 45.33,-105.39 45.33,-95.41 42.7,-77.21 42.3,-69.12 41.03,-61.27 38.9,-53.68 36.11,-45.08 36.11,-45.08 31.59,-37.7 27.73,-37.7 21.97,-43.14 13.13,-45.08 5.31,-47.28 -1.42,-47.58 -4.29,-48.34 -8.04,-55.59 -7.17,-57.85 -11.47,-66.69 -14.89,-73.73 -18.77,-80.54 -23.13,-87.14 L -33.02,-100.36 C -36.44,-104.92 -37.08,-107.74 -34.94,-108.84 Z M -34.94,-108.84" transform="rotate(-181.82)"  />
            <path id="pc_fan_cvs-fan_blade_3" fill-rule="evenodd" stroke="rgb(18, 18, 18)" stroke-width="1" stroke-miterlimit="4" fill="rgb(18, 18, 18)" d="M -34.94,-108.84 C -25.46,-114.04 -9.43,-116.66 13.13,-116.72 28.32,-116.16 38.18,-112.97 42.7,-107.14 45.33,-105.39 45.33,-95.41 42.7,-77.21 42.3,-69.12 41.03,-61.27 38.9,-53.68 36.11,-45.08 36.11,-45.08 31.59,-37.7 27.73,-37.7 21.97,-43.14 13.13,-45.08 5.31,-47.28 -1.42,-47.58 -4.29,-48.34 -8.04,-55.59 -7.17,-57.85 -11.47,-66.69 -14.89,-73.73 -18.77,-80.54 -23.13,-87.14 L -33.02,-100.36 C -36.44,-104.92 -37.08,-107.74 -34.94,-108.84 Z M -34.94,-108.84" transform="translate(1, 0) rotate(108.46)"  />
            <path id="pc_fan_cvs-fan_blade_4" fill-rule="evenodd" stroke="rgb(18, 18, 18)" stroke-width="1" stroke-miterlimit="4" fill="rgb(18, 18, 18)" d="M -34.94,-108.84 C -25.46,-114.04 -9.43,-116.66 13.13,-116.72 28.32,-116.16 38.18,-112.97 42.7,-107.14 45.33,-105.39 45.33,-95.41 42.7,-77.21 42.3,-69.12 41.03,-61.27 38.9,-53.68 36.11,-45.08 36.11,-45.08 31.59,-37.7 27.73,-37.7 21.97,-43.14 13.13,-45.08 5.31,-47.28 -1.42,-47.58 -4.29,-48.34 -8.04,-55.59 -7.17,-57.85 -11.47,-66.69 -14.89,-73.73 -18.77,-80.54 -23.13,-87.14 L -33.02,-100.36 C -36.44,-104.92 -37.08,-107.74 -34.94,-108.84 Z M -34.94,-108.84" transform="translate(1, 0) rotate(49.72)"  />
        </g>
    </g>
</svg>

```

Here is a demo of it in action:

[image]