wokwi/wokwi-features #137: option to connect multiple wires together (wire to wire connection)

matititam, 2021-07-06:
Example use case: 
[image]

In the above case, connecting the ground of all 10 LEDs to the Arduino UNO becomes tedious as well as takes time to make i look overlap one above the other so that it doesn't look tangled.

Wish is to be able to connect the wires together.. as shown in the below example:
[image]

djedu28, 2021-11-20:
Hello.
The feature is really interesting, especially for multiple connections in parallel.

A specific solution for your example (and that's what I use all the time), just connect each negative leg of the LED to the negative leg of the next LED;

follow the example:
https://wokwi.com/arduino/projects/315740099427959361

[image]

urish, 2021-12-12:
>  just connect each negative leg of the LED to the negative leg of the next LED

yes, that's a good workaround. I do this too :)

jaseinatl, 2022-09-21:
Though this visually works, it causes problems if you move stuff around a delete an LED. What might be a better solution is that  while you are creating a new connection wire, clicking on an existing wire would automatically extend your wire to its end point following the same path.

A junction like the first example, implies that irl there is a wire nut or terminal block or something. There are potential trouble locations that you would need to look at when troubleshooting if your circuit didn't work the way it was simulated.

But if in real life, you are running the wires to the same place in parallel along the same path, you need to be able to show that. By duplicating the path automatically, you can then move the elements around and select the wired connected to the individual elements separately. You could delete a component and not affect any of the other components that share the same terminal point.

The way I tried it (and it worked) was that I clicked on pin, then pulled the wire to an existing wire and clicked....but then I looked in the diagram and just pasted the rest of the connector's path.

What would also be great is being able to just click one terminal point, then the other and  a dynamic wire would be stuck to your pointer that you could stretch and drag wherever it looked the best. When you were done you could right-click to drop it or maybe his the space bar or something to say "drop it here". Having it snap to other wires would be a super easy way to do this as well.

urish, 2023-09-29:
Early prototype: 

https://github.com/wokwi/wokwi-features/assets/892318/b83a2f9a-18a5-4c4d-917a-3f6b87572456

shaos, 2024-01-23:
Not yet fully implemented?

urish, 2024-01-23:
Not yet - it has some edge cases with the ERC engine, and I haven't had a chance to look into them yet.

You can play with the prototype at https://junctions.preview.wokwi.com/projects/new/arduino-uno

urish, 2024-05-07:
Junctions are now available!

Example: https://wokwi.com/projects/377146682042662913

closed 2024-05-07 by urish

urish, 2024-05-08:
You add them through the add part ("+") menu:

[image]