wokwi/wokwi-features #138: observation report

matititam, 2021-07-06:
I observed the starting point of the wire to be off from the mouse pointer edge. 
The issue appeared in between. Earlier the wires were being drawn properly. once I deleted two wires and then tried to redraw the issue appeared. The screenshot is attached. Notice how soon after connecting the wire to Pin number one, the tip of the wire snatched back to the LED anode (rightmost). while moving the mouse, the wire stays a few lengths away. it happens to all the pions once it starts. 

The issues goes if I save the project/reload the project
[image]

urish, 2021-07-06:
Is it related to zooming / panning the diagram? A list of steps to reproduce will be very helpful!

urish, 2021-09-16:
Closing as I couldn't reproduce it. Feel free to reopen if you find a consistent way of reproducing the issue.

closed 2021-09-16 by urish