wokwi/wokwi-features #139: LED pin doesn't allow the connection to be made-sometimes

matititam, 2021-07-06:
Sometimes, the connection cannot be straightforward. here is a glimpse
[image]

urish, 2021-07-06:
Thanks for reporting! Which browser is this?

Can you send a link to the relevant project so I can reproduce it here?

matititam, 2021-07-07:
Link here: https://wokwi.com/arduino/projects/303372512437207617
Browser: Firefox
unable to recreate - will report soon if I found a way

matititam, 2021-07-07:
I just found a way to recreate it. I hope the below videos gives the hint!
[image]

I guess when an area of one device overlap with others, it will come up ( the thin green rectangle blocks/covers teh adjacet LED's Pins)

If I move the LED away, the problem will nto appear. Keeping LEDs closer was the root cause. I guess

urish, 2021-07-07:
> Keeping LEDs closer was the root cause

Indeed. Thanks for reporting, fixed!

closed 2021-07-07 by urish