wokwi/wokwi-features #143: feature request: wires bending option in simulation

matititam, 2021-07-10:
Can we have the option of smooth bending of wires? This is only to make the simulation wires look good. presently it is hard 90 degrees.

What I request is something like this for the wires to look like:

[image]

urish, 2021-11-14:
First preview of the rounded wire corners (along with a fairly complex diagram): 

https://round-corners.preview.wokwi.com/arduino/projects/309655684318757441

Before we merge, there are a few things we'll need to figure out:
1. Should we make the corners round by default?
2. What should the default corner radius be? 
3. Should we add an option to configure the corner rounding (enable/disable? set radius?)  
    If so, should it be part of diagram.json? Where does it fit?

Koepel, 2021-11-14:
1. Yes.
2. It's perfect.
3. No need.

There is a difference in the corners for the black and other colors. The black has rough pixels. When I draw a new black wire, then it is okay. It happens only with existing black wires. Five times enlarged:
[image]

matititam, 2021-11-15:
1. Yes
2. looks good/cant comment
3. Default option -shall be with bending corners. don't see the need for options to choose between bent and non-bent radius.

urish, 2021-11-15:
Thanks for the feedback! So I'll merge it after investigating how it might impact the performance of the diagram editor

urish, 2021-11-16:
Merged!

Re the rough pixels for black wires - wow, I'm impressed that you noticed that! I'm not sure if there's much I can do, it's how the browser renders these curves. How bad is it?

Koepel, 2021-11-16:
Not bad at all.
I only mentioned it in case the code has: `If color is black then render with a different method`

urish, 2021-11-16:
Perhaps Firefox does :P