wokwi/wokwi-features #146: Adding Protothreads library

naayem, 2021-07-15:
Hello, it will be great if we can add Protothreads library for clearer code implementation when elaborating complex project.

urish, 2021-07-15:
This library? https://gitlab.com/airbornemint/arduino-protothreads

naayem, 2021-07-15:
Hello urish, Yes! It's this library. Very practical. I didn't think of posting the link sorry. 
Thanks for reading me!

urish, 2021-07-15:
Alright, library added! 

What project are you currently working on?

closed 2021-07-15 by urish

naayem, 2021-07-15:
Waw that was fast 😄  I'm working on a regulation project where I want to mix components into a little reactor. Measuring the concentration through a captor and actuating the pump for regulation. Thanks yo your tool I can gain much time by simulating parts of code before having a real prototype.

Thanks a lot guys!

naayem, 2021-07-15:
I see that the website has been updated. I guess it will take some time to be updated on the servers to access the library in the simulation.

urish, 2021-07-16:
It should be available for you - is this not the case?

Also, I'm curious, how did you find out about Wokwi ?

naayem, 2021-07-19:
It is working thank you.

So I needed to simulate my outputs, to test and progress. The first program that pop up when searching is tinkercad.
But I was confronted to the impossibility to upload multifile code and it was a mess with libraries and the IDE at first glance. 
I searched for " Alternative to tinkercad multifile" and I found an article about wokwi. 
Now I am really happy to work with wokwi. At first there was an error status for no reason when you uploaded the library. I waited the next day and everything worked completely fine. Everything is going smoothly with multifiles and protothreads until now and that's really great.

urish, 2021-07-22:
Awesome :)