wokwi/wokwi-features #149: A Serial to Websockets device

Koepel, 2021-07-16:
Feature request: A special device with RX and TX pins that connects to a websocket. A Python3 script could be the websockets server and connect to a serial port on the computer.

If that is possible then a interface to a device (for example a scale with a serial port) can be developed from within Wokwi. It would also be possible to have a real Arduino board communicate with a simulated Arduino board.

urish, 2021-07-16:
What if the Serial Monitor RX/TX were available through a web socket? Would that also do the trick?

Koepel, 2021-07-16:
A serial duplex communication to the real world would be a dream come true. I hope you don't mind, but my feature request aims higher.

It is the same as connecting a Bluetooth or GPS module to pin 0 and 1 of a Arduino Uno. It works, but then a LCD display should be added to show messages.
Developing a interface to a serial scale depends heavily on the Serial Monitor.
Suppose a real Arduino sends real sensor data to a simulated Arduino. Then I would like to see the data in the Serial Monitor.

sutaburosu, 2021-07-16:
I think UART <-> websockets would be very useful.  I would definitely look at extending my Python proxy to be bi-directional and less focussed on transporting only LED data.

I think a library such as [VSync](https://github.com/ernestum/VSync) may be useful.  It provides a simple mechanism to sync variables between two sketches. It only supports `int`s currently.

urish, 2021-08-11:
Note: there's now an experimental support for connecting to an actual Serial port. Here's an example:

https://wokwi.com/arduino/projects/306534287811805760

Only works on Chrome / Edge (other browser do not support the Serial API, and it's not very likely to change in the foreseeable future).

You can also use a virtual serial port driver (such as [com0com](http://com0com.sourceforge.net/) on Windows) to make your Wokwi project speak to another software on your computer (or to a different wokwi sketch)

urish, 2021-11-16:
This one got high [in the feature list](https://wokwi.com/features), so it's a good time to start discussing the implementation.

One way would be to extend `wokwi-serial-port` (mentioned in the previous comment) to support web socket connection in addition to a real serial port, as a configuration option. 

For instance, we could add a "websocket" attribute that would point at the URL of the web socket. Something along the lines of:

```
    {
      "id": "serial",
      "type": "wokwi-serial-port",
      "attrs": { "url": "ws://localhost:1234", "baud": "115200" }
    }
```

Then, there's also the question of how we encode the data that goes into the socket. The most straightforward way would probably be to pass the binary data as-is, unmodified. We might need to introduce some buffering, as WebSocket are not stream-oriented. They break down the data into messages, and each message gets a 6 bytes header (client -> server). So sending every byte in a message would add a considerable overload, and might slow down the simulation (if we're running at 115,200, this would mean more than 10,000 messages per second).

urish, 2021-11-17:
> A device (such as the Logic Analyzer) could be connected to SoftwareSerial for example. Is the decoding of the serial signals possible ? What if the option to use multiple Arduino boards is implemented first and the websockets device uses parts of a Arduino Uno for the serial signals ?

It is possible indeed. You can also redirect the built-in Serial Monitor to different pins, and use SoftwareSerial on these pins. There's [a guide explaining how set up the Serial Monitor with ATtiny85 + SoftwareSerial](https://docs.wokwi.com/guides/serial-monitor#attiny85--softwareserial), but you can use the same process for the Arduino as well.

So for that part we're all set.

yepher, 2022-01-06:
My vote for the "special device" would be an FTDI (or CH340  etc) board. The USB port would represent the websocket interface. You could start with just defining the websocket encapsulation protocol and let folks implement their translation in their language of choice. Or just lean on one of the many existing solutions like [go-serial-websocket](https://github.com/ricorx7/go-serial-websocket). This way the simulator could talk to localhost or some publically accessible webserver with the physical port.

[image]

To test you can just connect to actual FTDI  TX->RX and RX-TX boards to your computer and have the serial console running on one and the other is the other one the server drives.

urish, 2023-01-04:
Note about the com0com workaround (from @jflamy): 

> you need to download the previous version 2.2.20 of com0com.  The later version doesn't work (signing issues).   Works fine with that version.

urish, 2023-02-17:
Note: this is now supported by [Wokwi for VS Code](https://marketplace.visualstudio.com/items?itemName=wokwi.wokwi-vscode). 

It uses the standard RFC2217 protocol, which is supported by the popular pySerial library. You can learn more [here](https://docs.wokwi.com/vscode/project-config#serial-port-forwarding)

mngyuan, 2023-04-12:
Is there a way to connect a Wokwi simulated Arduino's serial port to nodejs? Similar to the [com0com](http://com0com.sourceforge.net/) suggestion, but I'm on mac so I need something different. Why did implementation of this feature use RFC2217 instead of websockets? It seems there isn't a good RFC2217 node.js library.

jflamy, 2023-04-12:
> If you are only interested in textual data, then you can treat the RFC2217
> server as a standard TCP socket. If you need binary data, then you need to
> escape/unescape every occurrence of 0xff, but other than that, it's just a
> plain TCP socket.

On a Mac you would use socat I believe.

mngyuan, 2023-04-13:
ah ok, it's been a while since I've used sockets but I've tried the following with the simulator running with the `wokwi-serial-port` part in `diagram.json`:
```
$ nc -l 4000
nc: Address already in use
$ sudo lsof -i -P | grep :4000
Code\x20H 33993             liuyue   86u  IPv6 0xd070471690e4bf31      0t0    TCP *:4000 (LISTEN)
```
I'm not sure the simulator is properly outputting my `Serial.print()` calls - how do I see serial output from my Arduino sketch in the Wokwi for VSCode simulator and why is the simulator listening on port 4000 / preventing me from listening?

urish, 2023-04-13:
The simulator is opening a TCP server. You need to connect to it to see the prints.

mngyuan, 2023-04-13:
is that not what `nc -l 4000` does, listen on TCP port 4000?

urish, 2023-04-13:
Wokwi open a TCP server. You need to connect to it, localhost:4000 (or whatever port you set in Wokwi.toml), not listen on port 4000

mngyuan, 2023-04-13:
`nc localhost 4000` shows nothing on my system while the wokwi simulator is running.

urish, 2023-04-13:
What does your wokwi.toml look like? Are you using a VS Code dev container or WSL?

urish, 2023-04-13:
Looks good. Does your code keep printing stuff to the serial monitor? 

You'll only see what's been printed **after** netcat has connected.

afifkhoury2005, 2024-01-04:
I found the correct code to use in order to make com0com compatible with wokwi I recommend using the following code: {
  "version": 1,
  "author": "AFIF_CODE",
  "editor": "wokwi",
  "parts": [
    {
      "type": "board-esp32-devkit-c-v4",
      "id": "esp",
      "top": 0,
      "left": 0,
      "attrs": {}
    },
    {
      "id": "serial",
      "type": "wokwi-serial-port",
      "attrs": { "baud": "115200" }
    }
  ],
  "connections": [
    ["esp:TX", "$serialMonitor:RX", ""],
    ["esp:RX", "$serialMonitor:TX", ""],
    ["$serialMonitor:RX", "serial:RX", ""],
    ["$serialMonitor:TX", "serial:TX", ""]
  ],
  "serialMonitor": {
    "display": "always"
  }
}