wokwi/wokwi-features #152: LiquidCrystal_I2C library not compatible with ATtiny85

skeatz, 2021-07-31:
The LiquidCrystal_I2C library requires the Wire library (there's a #include <Wire.h> in the implementation file). The Wire library is not compatible with the ATtiny85, which means that simulation of a circuit with the ATtiny85 & I2C LCD display will not work. Wokwi reports an error message "/arduino/hardware/avr/1.8.3/libraries/Wire/src/utility/twi.c:88: 'SDA' undeclared".

Possible solutions are 1) add the LiquidCrystal_ATtiny library or Malpartida's NewLiquidCrystal library or 2) Modify the LiquidCrystal_I2C library to support the ATtiny85. Perhaps Malpartida's NewLiquidCrystal library could be added, as it supports both the Arduino Uno/Mega and ATtinys? Thank you.

urish, 2021-07-31:
How about the [Adafruit_LiquidCrystal library](https://github.com/adafruit/Adafruit_LiquidCrystal)?

skeatz, 2021-07-31:
The Adafruit_LiquidCrystal library is not implemented in wokwi. There is also the point that the Adafruit_LiquidCrystal library is based on the MCP23008 IO expander chip. Many of the lower cost I2C LCD modules use the PCF8547. I've tried the Adafruit_LiquidCrystal library on my physical I2C LCDs and have not been able to get it to work.

urish, 2021-07-31:
Thanks for the input! Is there any other version of that library that is available on the Arduino Library manager?

Alternatively, would you be interested to add LiquidCrystal_ATtiny  to the Arduino Library manager?

skeatz, 2021-08-01:
I've looked through the Arduino Library manager, found a few I2C LCD libraries. Most of them use the Wire library, so may not be compatible with the ATtiny85. Will need to investigate further.
I'll check out how to add the LiquidCrystal_ATtiny to the Arduino Library manager.

urish, 2021-08-01:
> I'll check out how to add the LiquidCrystal_ATtiny to the Arduino Library manager.

I did this recently with the [TinyDebug library](https://github.com/wokwi/TinyDebug). In a nutshell:

1. Create your own fork on GitHub
2. Add a library.properties file (and possibly also keywords.txt and some example as needed)
3. Create a release tag
4. Then open an pull request for [the library registry github](https://github.com/arduino/library-registry)

You can find the full instructions here: https://github.com/arduino/library-registry/blob/main/FAQ.md#submission

From my experience, the process is automated, so if the pull request passes their checks, it's merged immediately and the library becomes available within an hour or less.

different55, 2021-08-11:
Not sure if this makes a difference but SpenceKonde (aka Dr Azzy) has an ATTiny core that has built in support for Wire: https://github.com/SpenceKonde/ATTinyCore. Would be nice to be able to simulate that kind of functionality online.

urish, 2021-08-11:
> Would be nice to be able to simulate that kind of functionality online

So would you like this core to be added to the simulator? In that case, do you mind opening a dedicated issue for this request?

different55, 2021-08-11:
Certainly can. Didn't realize adding new cores was in the realm of possibility, but definitely glad it is!

urish, 2021-08-11:
Well, this is fixed by #157!

Example:

https://wokwi.com/arduino/projects/306656045893157440

closed 2021-08-11 by urish