wokwi/wokwi-features #153: shift register 74HC165

alto777, 2021-08-03:
I would like to use the 74HC165 shift register for input expansion.

I believe I would be competent to create this part; I hope it should be easy for someone who did this all time.

I haven't looked hard, I will. If you could shorten the circuit to where I might learn how to make, and submit for inclusion, additional parts for wokwi I would be greatful.

This thing can only get better.

TIA

alto777

urish, 2021-08-03:
Nice to meet you! I'll try to get the part done this week. I believe it should be pretty similar to the 74HC595, with some logic changes.

For the graphic elements, there's a [short video series](https://www.youtube.com/watch?v=gh27icNatwA) that walks through the creation of an element, as well as a [detailed blog post](https://www.smashingmagazine.com/2020/01/recreating-arduino-pushbutton-svg/).

What project are you working on? 

Also, I'm curious, how did you find out about Wokwi?

alto777, 2021-08-03:
OK, thanks and thanks.

Ppl were casually mentioning wokwi on the Arduino fora where I hang out to keep up with all the new things that are being done with these small computers. I took a look after the Nth time and was delighted by what I found.

I have no trouble lashing up hardware to test things, but this sim environment is the bomb, is that expression still in use?

Anyway, great work! I have been punching at the edges and have not found it to be wanting. Except, natch, the occasional odd-ball part. Nice selection already, seems like it would handle a great fraction of what turns people upside down with their various projects, LCD screen, neopixles, rotary encoder &c.

I like especially the idea of just dropping a link to a project into a thread. Easy for someone to run my idea for themselves.

I have no specific project right now, or rather way too many. I mostly have been hacking with smart pixels and animation, and gradually updating old projects for newer (cheaper) hardware. Now that AVR is owned by Microchip, or is it vice versa, I no longer feel disloyal. :-|

a7

urish, 2021-08-04:
Thank you so much for the feedback!

Behind Wokwi is a great community (you can meet most of the people on the [discord chat](https://wokwi.com/discord)), and @AriellaE and I are the primary developers. 

Mentioning Wokwi is a great help. We also have a [club](https://wokwi.com/club) for people who want to help us pay the bills :)

urish, 2021-08-04:
Meanwhile, I've added the 74HC165.

* Demo: https://wokwi.com/arduino/projects/306031380875182657
* Cascading demo: https://wokwi.com/arduino/projects/306024460940476993
* Docs: https://docs.wokwi.com/parts/wokwi-74hc165

alto777, 2021-08-05:
Nice. Fast work.

Suggest for symmetry that you add a note to the _output_ shift register part about the now available _input_ shift register!

a7

closed 2021-08-05 by alto777

urish, 2021-08-05:
Thanks Robert, just added as you suggested :)