wokwi/wokwi-features #158: Feature Request: basicMPU6050 library

different55, 2021-08-11:
Not sure how far we wanna go down the rabbit hole of adding multiple libraries that more or less do the same thing, but outside of rolling your own code, this is the smallest (in terms of flash size) library I've found in the Library Manager for just pulling off raw accel/gyro values off the MPU6050. If desired, it also supports correcting for gyro drift through some fancy math.

urish, 2021-08-11:
Library added!

closed 2021-08-11 by urish

different55, 2021-08-11:
Thank you much!

different55, 2021-08-12:
Just wanted to add that thanks to this library (and the others) being added, I was finally able to hunt down the last bug that's been keeping me from shrinking my Arduino project to ATTiny-scale, since the bug (missing call to Wire.begin in a library) that was plaguing me in the move to the ATTiny85 was also seen in the simulator, but *way* easier to pare down and isolate virtually. Thanks for helping me minimize the amount of hair pulling I was doing!

urish, 2021-08-12:
Amazing!