wokwi/wokwi-features #159: RGB LED with analogWrite() in Franzininho

arcostasi, 2021-08-12:
Please, the code below is not working in Franzininho

```cpp
int redPin = 4;
int greenPin = 1;
int bluePin = 0;

void setup() {
  pinMode(redPin, OUTPUT);
  pinMode(greenPin, OUTPUT);
  pinMode(bluePin, OUTPUT);
}

void loop() {
  colorRGB(255, 0, 0); // Red
  delay(1000);
  colorRGB(0, 255, 0); // Green
  delay(1000);
  colorRGB(0, 0, 255); // Blue
  delay(1000);
  colorRGB(255, 255, 125); // Raspberry
  delay(1000);
  colorRGB(0, 255, 255); // Cyan
  delay(1000);
  colorRGB(255, 0, 255); // Magenta
  delay(1000);
  colorRGB(255, 255, 0); // Yellow
  delay(1000);
  colorRGB(255, 255, 255); // White
  delay(1000);
}

void colorRGB(uint8_t red, uint8_t green, uint8_t blue) {
  analogWrite(redPin, red);
  analogWrite(greenPin, green);
  analogWrite(bluePin, blue);
}
```

with digitalWrite it worked normally like the example:
https://wokwi.com/arduino/projects/306670612160447042

urish, 2021-08-12:
I can confirm this is a bug in the simulator. Running the code on a physical ATtiny85 produces the following output (pins PB0/PB1):

[image]

zoomed-in, you can see the PWM pattern:

[image]

Running the same code in the simulator produces:

[image]

In both cases, the PWM wave looks very similar (2ms peak-to-peak / 500Hz, ~50% duty cycle), but it seems like `delay(1000);` stops working correctly while PWM is in use.

urish, 2021-08-12:
Debugging with GDB reveals an interesting phenomenon: `micros()` goes back in value!

Here's how I figured it out:

1. Started a WEB GDB session (debug mode)
2. Set a breakpoint
   `break delay`
3. Ignore the first breakpoint hit (so we actually break on the second, problematic call to `delay()`)
  `ignore 1 1`
3. Run the program
  `continue`
4. The program hits the breakpoint, added another printf breakpoint: 
   `dprint wiring.c:113,"MS=%d,start=%d,micros=%d,millis=%d\n", ms, start, micros(), millis()` 
5. Run the program
   `continue`

The observe the debug prints from GDB:
```
(gdb) c
Continuing.
MS=1000,start=1000576,micros=1001584,millis=1001
MS=999,start=1001576,micros=1004528,millis=1001
MS=998,start=1002576,micros=1004544,millis=1001
MS=997,start=1003576,micros=1004584,millis=1001
MS=996,start=1004576,micros=1003528,millis=1003
MS=995,start=1005576,micros=1003552,millis=1003
MS=994,start=1006576,micros=1003568,millis=1003
MS=993,start=1007576,micros=1003584,millis=1003
MS=992,start=1008576,micros=1003600,millis=1003
MS=991,start=1009576,micros=1003624,millis=1003
```

As you can see, `micros()` gets out of sync with `millis()` in the 2nd line (N=999), it jumps about 4ms forward, and then goes back in the 5th lines (N=996). This causes `delay(1000);` to count the time too fast and return prematurely.

Note: the first argument to `dprint` is the location of the following line in the source code for `delay()`: 

```
113                             ms--;
```

To find the line number, you can write `list delay,+10` to display the first 10 lines of `delay()`:

```
(gdb) list delay,+10
107     {
108             uint32_t start = micros();
109
110             while (ms > 0) {
111                     yield();
112                     while ( ms > 0 && (micros() - start) >= 1000) {
113                             ms--;
114                             start += 1000;
115                     }
116             }
117     }
```

Still need to hunt for the actual bug, but now at least we know where to look for it...

urish, 2021-08-12:
This seems to happen whenever the `micros()` implementation detects a timer overflow. You can observe it by adding another debug printf inside `micros()`:

```
dprint wiring.c:98,"Overflow!\n"
```

This corresponds to the line 98 inside micros:

```
(gdb) list micros,+23
81              uint8_t oldSREG = SREG, t;
82
83              cli();
84              m = timer0_overflow_count;
85      #if defined(TCNT0)
86              t = TCNT0;
87      #elif defined(TCNT0L)
88              t = TCNT0L;
89      #else
90              #error TIMER 0 not defined
91      #endif
92
93      #ifdef TIFR0
94              if ((TIFR0 & _BV(TOV0)) && (t < 255))
95                      m++;
96      #else
97              if ((TIFR & _BV(TOV0)) && (t < 255))
98                      m++;
99      #endif
100
101             SREG = oldSREG;
102
103             return ((m << 8) + t) * (64 / clockCyclesPerMicrosecond());
104     }```

If you repeat the instructions from the previous comment, and add this debug print right after the debug print in step 5 (the one inside delay()`, you'll get the following output:

```
(gdb) c
Continuing.
MS=1000,start=1000576,micros=1001584,millis=1001
Overflow!
MS=999,start=1001576,micros=1004528,millis=1001
Overflow!
MS=998,start=1002576,micros=1004544,millis=1001
Overflow!
Overflow!
Overflow!
MS=997,start=1003576,micros=1004584,millis=1001
Overflow!
Overflow!
Overflow!
Overflow!
...
Overflow!
MS=996,start=1004576,micros=1003528,millis=1003
MS=995,start=1005576,micros=1003552,millis=1003
MS=994,start=1006576,micros=1003568,millis=1003
MS=993,start=1007576,micros=1003584,millis=1003
MS=992,start=1008576,micros=1003600,millis=1003
MS=991,start=1009576,micros=1003624,millis=1003
MS=990,start=1010576,micros=1003640,millis=1003
MS=989,start=1011576,micros=1003656,millis=1003
MS=988,start=1012576,micros=1003680,millis=1003
```

So it seems like these "Overflow!" prints correspond with the times of discrepancy between `micros()` and `millis()`. In other words, the simulator probably incorrectly sets the `TOV0` bit in `TIFR` (or sets it correctly, but doesn't trigger the corresponding interrupt). Gotta keep investigating!

urish, 2021-08-12:
Issue found! 

The simulator is overriding the `TOV0` bit with the value of `OCF0A` because of a typo. Datasheet: 

[image]

Simulator:

```
const tinyTimer0Config = {
  ...
  // TIFR bits
  TOV: 1 << 1,
  OCFA: 1 << 1,
  OCFB: 1 << 3,
```

Should have been: `OCFA: 1 << 4`.

urish, 2021-08-12:
Fix is up!

closed 2021-08-12 by urish

arcostasi, 2021-08-12:
> Fix is up!

Thank you!

urish, 2021-08-19:
And now also a video with how I found this issue: https://youtu.be/YUUHw-bU9yk