wokwi/wokwi-features #160: ESP32/8266 wifi shield for my project

waltg72, 2021-08-16:
Can I request to have a ESP32/8266 wifi shield for my project. Thanks

urish, 2021-08-21:
Hi, what project are you working on?

waltg72, 2021-08-21:
Hi Uri,

I'm working on an IOT project where I can control the light, door, burglar
alarm, door bell of my miniture house project.

Below are the materials that I will be using for the project:

1. LED
2. Buzzer 1 - Burglar Alarm
3. ESP32 Wifi Shield (microcontroller)
4. Servo (Automated Door)
5. Buzzer 2 - Door Bell
6. smart Phone - To control everything
     through Wifi connection

I appreciate your help to make this happen starting with wokwi to visualize
and test first before building the actual project.

Thank you.

rgds,

Timothy

urish, 2021-08-21:
Thanks! The good news, Timothy - I'm working on ESP32 simulation. If you want to get early access to the preview version, you can send me a message on [discord](https://wokwi.com/discord).

waltg72, 2021-08-21:
Awesome, thanks!

urish, 2022-01-27:
Closing as we now have ESP32 with WiFi

closed 2022-01-27 by urish