wokwi/wokwi-features #166: Reed switch and magnet missing

Coldplay19, 2021-08-24:
Dear wokwi, 
can you please add a reed switch and a magnet for the simulation, please?

urish, 2021-08-24:
Hello, is there a specific model you are interested in?

Coldplay19, 2021-08-24:
any reed switch will do sir. Thank you so much for quick reply.

urish, 2021-08-24:
Thanks for the feedback! What kind of project are you working on?

Coldplay19, 2021-08-24:
actually i want to simulate a security alarm system

urish, 2021-08-24:
Would a standard slide switch be a good substitute for you until we have a proper reed switch?

Coldplay19, 2021-08-24:
when will i get the reed switch?

Coldplay19, 2021-08-24:
actually the slide switch won't . Sorry sir

urish, 2021-08-24:
Can't tell how long it'll take to add it. Creating the user interface part (in the [wokwi-elements](https://github.com/wokwi/wokwi-elements)) is the most time consuming parts, so once we have it there it should be pretty straightforward to integrate the actual switch into the simulation

Coldplay19, 2021-08-24:
thank you sir