wokwi/wokwi-features #169: ESP-01 WiFi modem (AT command)

MykolaKyrylenko, 2021-08-30:
Would it be possible to add the equivalent of an ESP-01 component for integrating to an Arduino Uno or Mega project at a WiFi modem. Some ideas:
- 4-pin interface
- USART interface (Rx and Tx only)
- 5V supply + Ground
- AT commands, which match the Espressif command set (https://docs.espressif.com/projects/esp-at/en/latest/AT_Command_Set/index.html).  A subset to enable a WiFi link and data transmission will be a great start.
- TCP/IP traffic sent via browser (not sure if this is possible)

Thanks.

urish, 2021-08-31:
Thanks for the suggestion @MykolaKyrylenko !

Would an ESP32 simulation also fit for your use case? Or do you need at ESP-01 specifically?

MykolaKyrylenko, 2021-08-31:
Thanks for the response.
I am flexible with the implementation, so an ESP32 will be good.
Are you intending an AT command interface?

MykolaKyrylenko, 2021-09-01:
I lecture at a technical college, and we are using the Atmel processor on the Arduino Mega to teach low level C programming, e.g. DDR, PORTC, etc. registers. The ESP32 will be too complicated for our students.
We are trying to incorporate IoT thinking, and I believe the simplest way it to have a "WiFi modem", with AT commands and a serial interface.
I would prefer not to change processors / boards part way through the topic.

NOTE: in industry I had to integrate a cellular modem to a microcontroller, and that worked well.

urish, 2021-09-11:
Got it. Yeah, ESP32 is indeed pretty complex.

There might be a way, at some point, to run both the AVR and ESP32 simulators in different tabs and have them communicate over virtual UART, which may be helpful. 

How long until you get to teaching the WiFi modem part?

MykolaKyrylenko, 2021-09-12:
Thanks for that.  Either solution will be great.
We are teaching the WiFi modem part in about 4 weeks, so I do not expect this feature this semester.  But we teach this topic each semester, so it will be great for the future.

PS I noticed another enhancement request, was using WebSockets.  Can something be done with that work?

MykolaKyrylenko, 2021-09-15:
I had an interim solution, if that is of interest.

Would it be possible to have a 'serial' device which just sends a HTTP POST request?
e.g. https://randomnerdtutorials.com/cloud-weather-station-esp32-esp8266/
[same hardware interface as my original post]

MykolaKyrylenko, 2021-09-16:
PS to my previous comment.  Sort of like the CURL command (https://curl.se/) to generate HTTP requests.

MykolaKyrylenko, 2021-10-08:
Thanks for the message.
Some of our students are remote, so doing your suggestion will confuse them.  For this purpose, I am getting them to use real hardware. It will be great to have something by next semester.

Overall, the students are loving Wokwi.  I needed to adapt bits due to known limitations, e.g. resistor divider network cannot be created for A/D conversion, so I used the potentiometer slider instead. A full integration of Wokwi and Falstad will be gold.
Great work.

urish, 2021-10-09:
Thanks for the feedback!

I opened a new issue (#203) for the resistor divider network, it's now also [open for vote](https://wokwi.com/features#feature-1021779688). Let's see how much love it gets!

urish, 2022-03-27:
This one is getting many votes recently, making its way to the top of the list. We may start working on it soon if it keeps coming up the list!

urish, 2022-04-24:
Some updates: I'm looking into using the official [ESP-AT Firmware](https://github.com/espressif/esp-at) on ESP32-C3 to provide this functionality. Anyone interested in beta testing?

MykolaKyrylenko, 2022-04-24:
I would be interested.  Do you think the ESP32-C3 can be included with an Arduino on the same project?

urish, 2022-04-25:
Thanks! In general, it is not possible to run two MCUs at once (at least until we tackle #186).

For the AT modem specifically, I'm working to make it possible. I'll share the details about the beta testing here in a few days.

urish, 2022-04-30:
So, here's something working - you can interact with the modem directly through the virtual serial monitor:

https://wokwi.com/projects/330317327704785492

The performance is pretty bad, and I haven't tried to connect it with an Uno yet (but in theory, that should also work). The only way to connect it is through diagram.json, as there's no UI element yet.

The WiFi works similar to how it does in the [ESP32](https://docs.wokwi.com/guides/esp32-wifi) - same access point configuration, uses the same gateway settings, etc.

ricardojlrufino, 2022-04-30:
WoW ... 

I tested it here, by the messages I saw that it is simulating very real...
Are you using real hardware on the server side??

[image]

urish, 2022-04-30:
Thanks @ricardojlrufino! Behind the scenes, we're simulating the [ESP-AT firmware](https://github.com/espressif/esp-at) using the same simulation engine that powers the ESP32 simulation. So it's not real hardware, but your questions shows that we are probably doing pretty good job with the simulation :-)

ricardojlrufino, 2022-04-30:
yes ...
failures are often good signs that things are working...  =]

urish, 2022-04-30:
Some more updates: the performance is now somewhat better (the simulation runs at 100% speed on my PC), and also fixed some issues with the UART output (wrong baud rate, missing chars when there's a lot of output, etc.).

So it should already be usable now. It will probably take some time until I get to create the board graphics, though.

MykolaKyrylenko, 2022-04-30:
Excellent work.  I had similar results to Ricardo.  Very fast performance.

I am assuming IP addresses are not being fed at the moment?
[image]

urish, 2022-04-30:
Thanks! This was a bug - can you please try again now?

MykolaKyrylenko, 2022-05-01:
Excellent:
[image]
AT+CWMODE=1 had instant resppnse
AT+CWLAP responded in a few seconds
AT+CWJAP repsonded with 'WIFI CONNECTED' instantly, and then after about several more seconds completed the response

urish, 2022-05-01:
Hooray :) 

If anyone gets a chance to test this with the Uno (or other Arduino board), I'd love to learn how it works

MykolaKyrylenko, 2022-05-03:
I tried it with the Mega using Serial 1. The code is very crude, but commands get sent and responses are received. Works well.
Responses appear on Serial Monitor.
See here: https://wokwi.com/projects/330589181982016082

urish, 2022-05-03:
Cool! I see you also tried to get an HTTP request working, any luck with that yet?

MykolaKyrylenko, 2022-05-03:
I managed to get a POST request received on our server:
[image]

The AT+HTTPCLIENT came back with an error:
[image]

urish, 2022-05-04:
Interesting, so it sent the POST request, but then said "ERROR" without telling anything else. 

Do you have a physical ESP chip to try it and see if you get a different result?

MykolaKyrylenko, 2022-05-04:
I have a physical ESP-01, and will try to get it going.

Some more information:
I sent a single POST request and 11 messages appeared on the server, with a 1 second gap between them.
[image]
Could be a timeout issue?

CURL command did as expected, with only 1 record sent:
[image]

urish, 2022-05-04:
Downloading a [PCAP packet capture](https://docs.wokwi.com/guides/esp32-wifi#viewing-wifi-traffic-with-wireshark) could be helpful to understanding what's going on

MykolaKyrylenko, 2022-05-04:
The 'Download WiFi Packet Capture (PCAP) file' option does not appear on my project.
Is this option only for ESP-32 projects?  Mine is Arduino Mega.

MykolaKyrylenko, 2022-05-06:
My server is a VPS on Binary Lane (https://www.binarylane.com.au/), and running standard Ubuntu + Apache.
I have not had time to plug in the ESP-01, but will attempt to soon.

Note: when I historically used the "WiFiEsp.h" library, it worked well.

MykolaKyrylenko, 2022-05-07:
My Espresiff firmwareis old
[image]
and does not have the AT+HTTPCLIENT command.
I will try to re-flash, or work our what command "WiFiEsp.h" library uses.

MykolaKyrylenko, 2022-05-07:
I tried with WiFiEsp library and no joy, even to initialise the ESP32-01
https://wokwi.com/projects/330956422342771284
I will keep on investigating...

MykolaKyrylenko, 2022-05-07:
Seems to work here without retries: https://wokwi.com/projects/330962510931821139

Note: intially, I tried to send the POST message with no delays between the characters, and I got a bad request error. By slowing down the data with 100msec delays (lines 84 - 91) the request was accepted without error.

urish, 2022-05-07:
Thanks for sharing your findings!

urish, 2022-05-12:
Update: The ESP32-01 now has graphics too, thanks to @AriellaE. 

The part name has changed to `board-esp-01`, and I have updated the example:

https://wokwi.com/projects/330317327704785492

Next steps: documentation, add the the part to the dropdown menu.

MykolaKyrylenko, 2022-05-12:
Excellent.
Looks good: https://wokwi.com/projects/330962510931821139

urish, 2022-05-14:
Lovely :)

rovoegit, 2023-07-31:
> I tried with WiFiEsp library and no joy, even to initialise the ESP32-01 https://wokwi.com/projects/330956422342771284 I will keep on investigating...

The newer library [WiFiEspAT](https://github.com/JAndrassy/WiFiEspAT) works with the ESP-01 Shield!   
--> https://wokwi.com/projects/371794654687376385   

I think it works because the wokwi ESP-01 shield has a newer firmware wich is supported by the new library.

CreativeQuantum, 2024-03-19:
Note:

In order for

```c
// disconnect persistent connection (not persistent)
WiFi.disconnect();

WiFi.begin("Wokwi-GUEST");
```

to work, you need to comment the following:

```c
#define WIFIESPAT1
```

in `WiFiEspAT/src/utility/EspAtDrvTypes.h`

gsingh93, 2024-03-31:
Thanks @CreativeQuantum, I was about to give up on using WiFiEspAt until I saw your comment (although now I see this information is in the README, I had missed it).

An easier way to do this, without modifying the library, is to add `-DWIFIESPAT2` to the build flags. For PlatformIO, that would look like `build_flags =-DWIFIESPAT2`.

CreativeQuantum, 2024-03-31:
Excellent work, glad there are other super heros fighting crime, huh, I mean, software complexity (which is a kind of crime of its own).