wokwi/wokwi-features #173: Small UX improvements

sutaburosu, 2021-09-01:
These are just possibilities to make the editor behaviour closer to "native" apps.

- [ ] When changing tabs in the text editor, give focus back to the editor if it had focus before.
- [ ] After creating a new file, switch to the new tab and focus the editor.
- [ ] When maximising/restoring the Simulation pane, keep the centre of view centred.

urish, 2021-09-01:
>  When maximising/restoring the Simulation pane, keep the centre of view centred.

How about when resizing the diagram? should we also keep the centre of the view centred?

i.e.

[image]

urish, 2021-09-01:
First two items done!

sutaburosu, 2021-09-02:
Nice!  Thanks.  Yes, I think centreing when resizing would be a good addition too.

closed 2021-09-02 by urish

urish, 2021-09-02:
And now also auto centring :-)