wokwi/wokwi-features #177: Grid snapping

sutaburosu, 2021-09-10:
Laying out diagram.json isn't easy, and expressing "busses" with wires routed neatly next to each other requires a calculator or fun mental arithmetic.

I feel there would be value in having grid-snapping.  So we might have a button toggleable between 3 or more states. I envisage 0.1" / 0.05" being the most useful,  which translates to multiples of 7.2 / 3.6 CSS units.

urish, 2021-09-12:
Added initial support for grid-snapping - 0.1" grid. When routing wires, press the Shift key to snap.

sutaburosu, 2021-09-12:
Thanks!  This is great so far.  The grid seems to be relative to the source pin.  Given the elements' pins may not align with a global grid, I think this is a good compromise.  It's certainly a lot less work to make a pretty diagram now, even if nudging slightly elements is necessary.

urish, 2021-09-12:
One work around I was thinking about is snapping elements to grid based on their first pin when you shift-drag them. For most elements this would probably do the trick

urish, 2021-09-20:
Experimental snapping implementation by @TimMathias: 

https://wokwi-diagram.vercel.app/arduino/new?template=arduino-uno

> There are two grid sizes: coarse (2.54 mm or 0.1") and fine (1.27 mm or 0.05"). Parts are snapped to the coarse grid only; new wires are snapped to the coarse grid by default, or the fine grid by pressing Shift, or off grid by pressing Alt.

Please share your feedback here