wokwi/wokwi-features #178: MAX7219 Part Request

rcwilcox, 2021-09-11:
Would it be possible to add the MAX7219 Seven Segment Decoder?  I would like to use this to prototype/simulate projects for STEM education using Multiplexed Seven Segment LEDs.   

**[MAX7219 Datasheet](https://datasheets.maximintegrated.com/en/ds/MAX7219-MAX7221.pdf)**

sutaburosu, 2021-09-11:
@rcwilcox if you have an 8-digit 7-segment display with a MAX7219, could you please check which order the digits are wired?  To my mind, it would make most sense for digit 0 to be on the right, increasing to digit 7 on the left. It would be nice to confirm this on real-world hardware.

rcwilcox, 2021-09-11:
@sutaburosu I haven't prototyped this with real-world H/W since I don't have the H/W.  I 'simulated' the interface (Address/Data) on Wokwi with two 74HC595 Shift registers chained together over SPI and used Digit 0 (labeled sevseg:DIG4 on a 4 segment wokwi-7segment part) on the right.  Of course I had to do the hex->segment decode in S/W for this to work.  This indeed make the most sense.

sutaburosu, 2021-09-11:
I found [this datasheet](https://www.parallax.com/package/max7219-8-digit-display-driver-datasheet/) for an 8-digit display. It's still not entirely clear to me which end is digit 0. The only hint I can find is at the bottom of page 9 where they mention blanking leading 0s using the scan-limit register. This tends to suggest digit 0 is on the right.

rcwilcox, 2021-09-11:
Wouldn't it depend on your implementation of the display with the 7219?  To be clear, I am looking for the 7219 part alone, not 7 segment LED(s) integrated with a 7219 (which seems to be what the 7219-LED matrix is).  You would use it as a decoder between the serial interface and the standard 7 segment display.

sutaburosu, 2021-09-11:
Oh!  I was imagining a part similar to [the existing MAX7219 64 LED matrix](https://docs.wokwi.com/parts/wokwi-max7219-matrix), but with a configurable number of 7-segment digits. I can see that a MAX7219 is a useful component in its own right, if you're keen to show the wiring.

rcwilcox, 2021-09-11:
This is how I simulated it

[image]

rcwilcox, 2021-09-11:
Timing diagram
[image]

rcwilcox, 2021-09-11:
Here is the API.  Note that I had to pre-decode the Addr and Data in the AVR software to set the DIGn Enable (Address) and y Segment Data.  The 74HC595 part would simulate this decoding per the datasheet.
[image]

urish, 2022-05-07:
Closing this issue as it hasn't received any votes in 6 months. Note that this could now also be implemented as a custom chip (see #301)

closed 2022-05-07 by urish