wokwi/wokwi-features #180: Transistors

paarangat, 2021-09-12:
Would it be possible to add npn transistors to Wokwi? They are really useful for the creation of electronic switches.

urish, 2021-09-12:
Thanks for the suggestions! Do you have a specific project in mind?

paarangat, 2021-09-15:
Sorry I found out my solution. Either ways, thanks!

urish, 2022-02-17:
Thanks for the suggestion!

One major difference between relays and transistors are their response times (millis vs nanos). So with transistors, you'd be able to implement an oscillator that will crawl the simulation down to halt. 

But I do see where they come into play when controlling LEDs and Relays. Other than the oscillator issue and the fact there's no real analog simulation, I think they shouldn't be too hard to implement.

leftCoast, 2022-02-17:
This sure looks like a slippery slope to me..

Koepel, 2022-02-17:
It will generate problems and we will have to explain that it is only a digital simulation. Could something be added to the symbol ? Perhaps a gray symbol ?

urish, 2022-02-17:
Gray symbol? What would it look like?

Koepel, 2022-02-17:
With the two colors and the dotted circle, it shows that there is something special about it.
[image]

Phetdanai, 2023-12-22:
So, how can I insert it in wokwi by 'diagram.json'?

Phetdanai, 2023-12-22:
Please type code below.

Phetdanai, 2023-12-22:
> With the two colors and the dotted circle, it shows that there is something special about it. [image]

Ok

oscar-unj, 2024-02-14:
It is very Important to add Transistors Llike BJT, FET, MOSTFETs Etc, Optocoupler, CI 555, Bare Atmega328p, OPAMs, Etc. like PROTEUS to have an Excelent Simulator.

dah78, 2025-05-17:
They are really useful for simulation of electronic switches

nashjb, 2025-07-07:
**Hey guys,**

Below is a simple implementation of a **MOSFET-N** using the Wokwi custom chip feature.

I feel like it is a fundamental part and that a MOSFET should be included by default. I don’t have access to the core Wokwi codebase, but I was wondering if this could be added as an official part (after a look over)?

### `mosfet-n.c`

```c
// Digital N-MOSFET (low-side switch) for Wokwi
#include "wokwi-api.h"
#include <stdlib.h>
#include <stdio.h>

typedef struct {
  pin_t gate;      // Gate
  pin_t drain;     // Drain
  pin_t source;    // Source (tie to GND)
} chip_state_t;

static void gate_changed(void *user_data, pin_t pin, uint32_t level) {
  chip_state_t *m = (chip_state_t *)user_data;

  if (level) {
    // Gate HIGH → MOSFET ON → Drain pulled to Source (GND)
    pin_write(m->drain, pin_read(m->source));
  } else {
    // Gate LOW → MOSFET OFF → Drain pulled HIGH (simulate disconnection)
    pin_write(m->drain, HIGH);
  }
}

void chip_init(void) {
  chip_state_t *m = malloc(sizeof(chip_state_t));
  m->gate   = pin_init("GATE", INPUT);
  m->drain  = pin_init("DRAIN", OUTPUT);
  m->source = pin_init("SOURCE", INPUT);

  if (pin_read(m->gate)) {
    pin_write(m->drain, pin_read(m->source));
  } else {
    pin_write(m->drain, HIGH);
  }

  const pin_watch_config_t cfg = {
    .edge = BOTH,
    .pin_change = gate_changed,
    .user_data = m,
  };
  pin_watch(m->gate, &cfg);

  printf("MOSFET-N initialized (low-side switch)\n");
}
```

---

### `mosfet-n.json`

```json
{
  "name": "MOSFET-N",
  "author": "Your Name",
  "pins": [
    "GATE",
    "DRAIN",
    "SOURCE",
    "",
    "",
    ""
  ],
  "chip": "mosfet-n.c"
}
```