wokwi/wokwi-features #181: Multiple .INO files are handled differently than the Arduino IDE

sutaburosu, 2021-09-12:
The Arduino build tools have special handling for .INO files:  function prototypes are automatically generated and inserted into the "main" .INO.  The "main" .INO is named the same as the folder the sketch is in.

In Wokwi we can already add additional .INOs, but we are met with compilation errors due to the missing prototypes.

As we discussed privately, it seems relatively simple to mimic the Arduino IDE's behaviour with multiple INOs.

[This test sketch](https://wokwi.com/arduino/projects/309546680351457858) currently fails on Wokwi, but works with arduino-cli.

urish, 2021-09-12:
Should now work. Can you please test and report back?

sutaburosu, 2021-09-12:
Seems to work perfectly. Thank you very much!

closed 2021-09-12 by sutaburosu