wokwi/wokwi-features #184: Ghosting? while dragging the recently selected element

matititam, 2021-09-18:
Steps to recreate (not sure exactly)

1. Open template [project](https://wokwi.com/arduino/new?template=arduino-uno)
2. place an LED
3. click on the Arduino UNO board
4. Try to move the LED now, by clicking on the LED and moving it
5. Sometimes, the ghosting will appear, when I try to drag the LED downward
6. I think, it is a combination of mouse 'pointing' to the Arduino pins momentarily as well. 
[image]

urish, 2021-09-18:
Thanks for reporting! Which browser?

matititam, 2021-09-19:
The browser was firefox.

sutaburosu, 2021-09-19:
I've tried to reproduce this. Once, I made it happen with the mouse alone. It seems difficult to trigger. I think it may be due to the pin name tooltip popping up, and the drag starting simultaneously. The tooltip text doesn't need to be selectable; disabling that may make a difference.

I did find a method which reliably produces similar results in both Chromium and Firefox: press ctrl-A (select all text on page) before dragging the LED. Every element I tested can trigger this, except the MCU which appears to be immune.

urish, 2021-09-21:
> except the MCU which appears to be immune

I wonder why... Perhaps it has to do with the fact that the MCU has some text on it?

> The tooltip text doesn't need to be selectable

Doesn't seem to make a difference for the Ctrl+A scenario. The reason I originally made it selectable is to allow copying the names of the pins so you can paste in the "connections" section of diagram.json. But now that we have mouse wiring, that's less of a concern.

urish, 2022-06-14:
Closing at this due not easily reproduce. Ctrl+A does make it reproduce, but I could find a quick way of fixing it.

In any case, I have made the pin name tooltip unselectable, as I find that sometimes it gets in the way while drawing wires.

closed 2022-06-14 by urish