wokwi/wokwi-features #185: Add capability to Connect Logic Analyzer as a Device in PulseView

rcwilcox, 2021-09-21:
Using PulseView to analyze .vcd files created by the Logic Analyzer requires:

1. Stopping simulation to write the .vcd file
2. Loading .vcd file into PulseView
3. Loading config file (as needed depending on use)
4. Periodically deleting the old files that are no longer needed.

While some of this can be mitigated via file associations, would it be possible to set up the Logic Analyzer as a Connected Device in PulseView?

urish, 2022-03-26:
Closing as this issue hasn't received any votes in 6 months

urish, 2022-03-30:
Reopening as the issue just got some votes

Gebhartj, 2024-02-09:
Libsigrok can be connected to a device (also in PulseView).

It should be possible to modify the [demo](https://github.com/sigrokproject/libsigrok/tree/b503d24cdf56abf8c0d66d438ccac28969f01670/src/hardware/demo) device to wait for a *.vcd file in some location to change and then stream data from that file.

Or... Implement some TCP/IP protocol, for example, [rigol ds](https://github.com/sigrokproject/libsigrok/tree/b503d24cdf56abf8c0d66d438ccac28969f01670/src/hardware/rigol-ds) to work with Wokwi.

The first option should be doable independently, but it adds little or no user comfort. The second option requires support from Wokwi, but I don't know if it's possible to contribute or just wait and spend some vote powers.