wokwi/wokwi-features #188: Request for delete on iPhone/iPad

TimMathias, 2021-09-21:
There seems to be no way to delete a part on iPhone/iPad through the UI. Does this issue also affect smartphones and tablets in general? e.g. Android.

Perhaps, when selecting a part, an "X" could be displayed in the top right corner of the part similar to the way the "?" is displayed for the help documentation.

Or perhaps a context menu could be created with standard options such as Cut, Copy, Paste, Delete, Rotate and Properties. The properties box could display options pertaining to specific parts like colours and text for LEDs/buttons, etc.

rcwilcox, 2021-09-22:
I couldn't find a way either on iPhone today.  I like the "X" idea, though it should probably have a confirm dialog, especially on a physically cramped UI.

urish, 2026-10-27:
Thanks for bumping this and for the detailed list of parts. I pushed a fix and the toolbar with rotate, duplicate and delete, should now show for every part in the diagram.