wokwi/wokwi-features #189: More compiler output OR show me what I am missing.

alto777, 2021-09-22:
I've developed a habit of having a throw-away real project open so that when I get the dreaded "Build Failed", yes, believe it or not it has happened. :-|

Anyway so that - I can cut and paste all code from the wokwi into the IDE and "Verify" it to get some (somewhat) better diagnostics.

Presumably at some point all that is suppressed in the wokwi, even a kludge like an "error.txt" file which if present would get some verbose output.

Priority low - the technique I employ is now second nature. Just wondering if there is a setting in the documentation which I promise you I have not scoured.

TIA

a7

sutaburosu, 2021-09-22:
In my experience, any compilation errors are presented in a red box in the editor pane. Use the arrows in that box to switch between the individual errors. If the error is on a different tab, you may have to switch to that tab and build again for the errors to be presented.

Sometimes it can be easier just to open your browser's Javascript console. All the build errors are printed there.

urish, 2021-09-23:
Related: #1. I'm closing this and merging the request with #1.

closed 2021-09-23 by urish

alto777, 2021-09-23:
Thank you @sutaburosu, I will see if that is all and what I am missing.

a7