wokwi/wokwi-features #190: Include ds1307 library

M-square, 2021-09-24:
The simulator has the DS1307 RTC chip as component - but not the DS3231
The simultator can include the DS3231.h library but not the DS1307.h.

The DS1307 can be used with the DS3231 library, but the functions differ in name, parameters etc. For consistency it would be nice if library and part match. My preference is the DS1307 - I use it in real projects. My current workaround is to create a wrapper-class that looks like RTC of DS1307 but calls the DS3231-library for the functuions I need. This is so my wokwi test code does not need lots of edit when transported back to a physical circuit.

urish, 2021-09-24:
Thanks for the request! Have yo trying including the library by adding a [libraries.txt file](https://docs.wokwi.com/guides/libraries) to your project?

M-square, 2021-09-24:
#include <DS3231.h>

class RTC_ {  // actually I use the DS1307 so I need a wrapper 
  public: RTC_() ;
  void getTime() ;
  int hour ;
  int minute ;
  private: DS3231 clock;
  bool Z ;
}  ;

RTC_::RTC_() { Z=false ;} 
void RTC_::getTime() {
  hour = clock.getHour(Z,Z) ;
  minute = clock.getMinute() ;
}

void setup() {
  Serial.begin(9600) ;
}

void loop() {
  Serial.println(millis()) ;
  PrintTime() ;
  delay(1234) ;

}

/*----*/void PrintTime() /*----------*\
|  Output RTC variable to Serial      |
\*---------------------------------- */
{
  byte Hour, Min  ;
  RTC.getTime() ; 
  Serial.print('[');
  Hour = RTC.hour%12 ;
  Min  = RTC.minute ;
  if (Hour<10) Serial.print('0');Serial.print(Hour);Serial.print(':');
  if (Min <10) Serial.print('0');Serial.print(Min );Serial.print(']');
}

--
I also tried using #include <RTClib.h> as in the RTC example https://wokwi.com/arduino/libraries/RTClib/ds1307 but that, too uses a different structure of functions (to do the same thing)

M-square, 2021-09-24:
Noticed the intelligent cut-n-paste seemed to have messed up comments around the PrintTime() function. Should be:
/*----*/void PrintTime() /*----------*\
| Output RTC variable to Serial |
*---------------------------------- */
(It looks nice in monofont)

matititam, 2021-09-24:
Hi
I can help you with the inclusion of required library in your project for now. Shall we connect in discord? https://wokwi.com/discord

M-square, 2021-09-24:
@matititam Yeah - sure! Msquare#3384

M-square, 2021-09-24:
In preparing myself for discussion I discovered
1) My ds1307.h is very very old. I have even patched my copy to cope with new calls/includes to wire.h.
2) Libraries.txt does not find anything, nor does the Arduino library manager. Or rather, it finds newer vesions.

I withdraw the issue. ('ll use the wrapper class in my projects on wokwi)

closed 2021-09-24 by M-square