wokwi/wokwi-features #191: Addition of Stepper Motors as devices

M-square, 2021-09-24:
I usually control stepper motors. These seem to be missing in Wokwi. I can see a few challenges to implement them

1: Visual apperance - Similar to the Servo, with a shaft where the angular position is visible, like an arrow glued on.
2: Angular position - accuracy/resolution. Steppers can have lots of steps, a small readout of the actual step it is on would be nice. Can be in the visual model or mouseover display.
2a: partial steps - steppers can be halfstepped so such a readout should be with decimal. (Implementation of detecting halfstep might be tricky)
3: Stepper types - bipolar (4 wire), unipolar(5 or 6 wire), are most common(?) but there are several. That means at least two fundamental different devices, in how they react to electrical connections. The visual representation might be different, possibly weakly indicating 2 bipolar coils or 4 unipolar coils.
4: Stepper resolution - An attribute is number of steps/revolution.
5: Gearing, and other mechanical - must be ignored, or you need to write a physics simulator, too. A geared stepper would simply have more steps to complete a revolution.
5a Possibly exception, but as a new device - Stepper with a linear actuator. (like potentiometer and slide potentiometer)
6: Electrical - in the simplified Wokwi simulation the Arduino drives coils directly - no backemf diodes, no transistor/FET, no 12V, no overcurrent etc. Nor do they have limits on steps/second and other RealLife trivia.
7: Driverchips - usually (bipolar) steppers are driven with a chip. In essence they have a Direction and Step input. A suggestion is a Stepper-with-driver device, where the chip is "integrated" (visually as a black pimple on side :-) ) Input lines are the Step, Direction (and possibly Enable ...) 
7a: Where chips offer half/quarter steps, one simply uses double/quadruple the step/revolution of the stepper as a simulation.

Simplification: Steppers need only step on transition when coils get correct +/- (fullstep) compared to previous state.

Oh ... and then there are the libraries for Steppers, with/without chips.

M-square, 2021-09-24:
Of course I am willing and able to help with a few of the implementation steps.

urish, 2021-09-24:
Great summary, thanks! I added this issue to the list of [features open for voting](https://wokwi.com/features), let's see if it gets any votes

M-square, 2021-09-25:
Something like this.
[image] [image]

urish, 2021-09-25:
Perhaps it'll also be useful to include some text indicating the exact angle?

M-square, 2021-09-26:
See point 2. Is angle or step units the more relevant? (Depends on what the code is trying to express, what you want to debug) Is an angle in radians, degrees or grad? All are possible options, and maybe they can be controlled by an attribute, as well as on/off of the text. I have also speculated on showing "speed", but thought that is a RealLife problem that the (relative simple) simulation need not reflect.

NB: Should the arrow+shaft be a group so they can be animated as a unit?
NB2: is there a need for some "hidden" small squares that can act as the connecting point for the wire?

DavidThijs, 2021-11-07:
Steppers would be awesome but it also requires support for the drivers (eg. TMCxxxx,A4988,DRV8825, or H-bridges like the L298)

M-square, 2021-11-08:
The point has been adressed in (6) and (7) in the original post

The Woki is not an electrical simulator, so it really doesn't matter which driver board. Neither do we need a 12V supply etc. There are two types: The ones with a DIR and STEP input, and those with 4 inputs to control two H bridges. The ULN2003 typically used for small unipolar steppers can be skipped in the simulation (sort of like the resistor for a LED). The Microstepping supplied by some driver chips can be effectivly simulated by setting the stepsize of the stepper - at the end of the day, the code in the arduino just sends step pulses and it notches the stepper onwards by n-degrees.
Thus, a stepper with (f.ex) a DRV8825 allready connected would be ok - it just exposes pins DIR and STEP (and enable). And likewise for one with L298 (4 in-pins, two enable). 

My reason for the original requst post is to provide a simple visual indication of how much the stepper has been turned by the program, where Serial output is too messy.

urish, 2021-12-17:
Update: we have the Wokwi Element for the stepper motor: 

https://elements.wokwi.com/?path=/story/stepper-motor--default&args=angle:45

urish, 2022-01-09:
This part has been getting many votes recently, so I'll be starting to work on the actual implementation soon

urish, 2022-03-27:
Initial implementation: https://wokwi.com/projects/327324886912467538

I'm keeping this issue open until we also have the docs ready.

The implementation also supports half-stepping (though the Arduino Stepper library doesn't).

This works without a driver, just Arduino driving the coils directly. Would love to get feedback + if you would like to see any driver. please reply with the model name.

p.s. a very good video for understanding how stepper motors work: https://www.youtube.com/watch?v=eyqwLiowZiU

francisduvivier, 2022-03-27:
Awesome that we are getting stepper support! Here a vote for the A4988
and/or TB6600 driver.

Grts,
Francis

M-square, 2022-03-28:
YES! **+1** thumbs up etc. I wont repeat the features I desired (demand? ;-) ) they are in the first post. Where is the preliminary doc/source so I can see what attributes are beta/available now? I made a simpler demo - not using libraries : https://wokwi.com/projects/327360139702043220 (It seems non intutive what the start position is, ie. which coil combination would not cause a step at start)

urish, 2022-03-28:
Thanks for the feedback + sample!

There are no docs yet - just got the implementation working. Docs are next. But there are also no attributes yet - are there any attributes you think we should add?

The start position is the same as A coil driven positive, B coil driven negative, as this is the start position that the Stepper library assumes.

M-square, 2022-03-28:
Attributes: 
1) Steps/revolution [4-4000] (the high number would simulate a geared stepper)
2) Position indicator: 
 (i) None(ie just the D-shaft) Steps (like now)
 (ii) Angle (in degrees)
 (iii) Arrow (a large visual arrow attached to shaft. Also it gives an indication of sweepspeed and if there is much motion one can not read the flickering numeric display but get a good idea from the arrow)
 (iv) Hour/minute - the angual readout is as a clockface would be - OK, that is just "For extra credit" ;-)

I really see several models - with and without driver chips. It is just "noise" on the diagram to connect the driver to the stepper, that is, Wokwi steppers have the driverchip allready attached/integrated.
a) Step/Dir style controller (DRV8825,  A4988 and TB6600 are that. Subtle differences with partial steps. Debatable if Sleep/Enable modes need any implementation behind the connection)
b) 4 wire H-bridges (like L298)
c) Direct 4 wire stepper - like now.
d) Direct 5/6 wire unipolar steppers (ULN2003 is just a one-to-one amplifier, eyecandy without function in the Wokwi)

I do not know enough about the underlying structure to be opiniated if "Model" can be an attribute, or each is a seperate part.

drf5n, 2022-03-28:
Here's an example using the AccelStepper.h library, a slide pot, and the README.md file for a Description tab that points back here:  https://wokwi.com/projects/327381547863769683

urish, 2022-03-28:
Thanks @M-square !

> (iv) Arrow (a large visual arrow attached to shaft. Also it gives an indication of sweepspeed and if there is much motion one can not read the flickering numeric display but get a good idea from the arrow)

Mind sketching this option?

M-square, 2022-03-28:
> Thanks @M-square !
> 
> > (iv) Arrow (a large visual arrow attached to shaft. Also it gives an indication of sweepspeed and if there is much motion one can not read the flickering numeric display but get a good idea from the arrow)
> 
> Mind sketching this option?

https://github.com/wokwi/wokwi-features/issues/191#issuecomment-927127775

drf5n, 2022-03-28:
> Attributes:
> 
> 1. Steps/revolution [4-4000] (the high number would simulate a geared stepper)
> 2. Position indicator:
>    (i) None(ie just the D-shaft)
>    (ii) Steps (like now)
>    (iii) Angle (in degrees)
>    (iv) Arrow (a large visual arrow attached to shaft. Also it gives an indication of sweepspeed and if there is much motion one can not read the flickering numeric display but get a good idea from the arrow)
>    (v) Hour/minute - the angle readout is as a clockface would be - OK, that is just "For extra credit" ;-)
> 
> I really see several models - with and without driver chips. It is just "noise" on the diagram to connect the driver to the stepper, that is, Wokwi steppers have the driverchip allready attached/integrated. a) Step/Dir style controller (DRV8825, A4988 and TB6600 are that. Subtle differences with partial steps. Debatable if Sleep/Enable modes need any implementation behind the connection) b) 4 wire H-bridges (like L298) c) Direct 4 wire stepper - like now. d) Direct 5/6 wire unipolar steppers (ULN2003 is just a one-to-one amplifier, eyecandy without function in the Wokwi)
> 
> I do not know enough about the underlying structure to be opiniated if "Model" can be an attribute, or each is a seperate part.

+1 on the steps/rev.  Those can be very different dependent on gearing etc.

Maybe an arrow-length attribute, with 0/none as the default.  And an initial position attribute. taken together, those are a very configurable dial.  When playing with real steppers I use a bit of tape or a twist of wire to see how they work together

A generic step-dir-enable type driver would be good.  I'm not sure how many specific divers would be good to support.  adjustable microstepping, serial control, etc... seem like a combinatorial explosion of possibilities, while a generic step+dir option and steps-per-rev attribute could serve many use-cases.

urish, 2022-03-29:
> https://github.com/wokwi/wokwi-features/issues/191#issuecomment-927127775 (The SVG files were uploaded at the time...)

Sorry for missing this part, and thanks for the sketch!

So here's a proposal for what the attributes could look like, would love to have your feedback:

## Attributes

| Name      | Description                                                                                          | Default value |
| --------- | ---------------------------------------------------------------------------------------------------- | ------------- |
| arrow     | Displays an arrow to indicate the position of the stepper.                                           | ""            |
| gearRatio | The gear ratio of the motor. set to "1" for 200 steps/revolution, "2" for 400 steps/revolution, etc. | "1"           |
| display   | What number to display on the stepper? Valid values "steps", "angle" (in degrees), "none"             | "steps"       |

Also, adding a fun sketch by @koepel:

https://wokwi.com/projects/327424914940232274

Using relays to control the stepper without any MCU code!

M-square, 2022-03-29:
> #191 (comment)

Re "gearRatio" : I have steppers with 24 steps/revolution, 2048 and 2047.6
(the latter two with gearbox), apart from the usual 150, 200, 400. So why
not just a simple integer for steps per revolution? Well, gear ratios can
be "weird" - they are usually given as 5:18 or something like that (as
gears by nature have to be integers) rather than 0.27 as the fraction is a
recurring decimal. In the code you do (integer) multiplication before the
division, and your rounding will come out right. In short gearRatio is
better, with at least floats but but better if a ratio (all paras are
strings anyhow, so ou only need to test if there is a ":" in it and convert
appropriately)
M²

urish, 2022-03-29:
Not sure if I understood correctly: You are saying gearRatio is better than steps/rev, but we should support "n:m" format in addition to decimal?

M-square, 2022-03-29:
Well, you wrote gearRatio as the attribute to describe the "granularity" of the stepper, with the underlying statement that it was relative to 200step/revolution. And that is fine. It's just a documentation note that 1=200steps. If Steps is an ALTERNATIVE to specifying gearRatio that fine, too, maybe better. My point was that 1, 2 as possible values is not enough, you need 0.5, 3.2 etc as well. And unfortunately gear ratios are not nice decimals numbers so therefore the suggestion that a ratio or fraction is allowed, too, as that can be represented and calculated with exactly.  My 48 stepmotor needs a (fake) gearRatio of 4.166666 or "200:48" (Whether that needs to be two attributes or you can parse the field conditionally on a ":" being present is out of my scope ) I admit for simulation purposes, 4.1666 probably is enough precision for the visual effect. It is quite critcal though that 48 pulses do end up as one rotation, without rounding effects piling up after a number of full turns. I at least, am always chasing the "off by one" errors in software, so I'd look closely if I did do a full rotation.

You're doing this in your (free?) time so I'm happy with anything - it's a question of how hightech/perfect it should be.

urish, 2022-03-29:
How about the following explanation for gearRatio?

| Name      | Description                                                                                          | Default value |
| --------- | ---------------------------------------------------------------------------------------------------- | ------------- |
| gearRatio | The gear ratio of the motor. set to "1:1" for 200 steps/revolution, "2:1" for 400 steps/revolution, etc. | "1:1"           |

M-square, 2022-03-30:
That gets my thumbs up vote.

urish, 2022-03-30:
Alright, so we now have both `display` and `gearRatio`.

`arrow` coming up next!

urish, 2022-03-30:
arrow done!

Also, docs page: https://docs.wokwi.com/parts/wokwi-stepper-motor

And the stepper motor is now available from the "Add part" menu

M-square, 2022-03-31:
The step/angle display is wrong, IMHO. If I fire 100 pulses to the 200 steps/rev stepper, I expect it to turn half a revolution, the display reads 100 steps, the angle is 180 degrees. If the stepper has a 1:4 gearing that would be equivalent to a stepper with 800 steps/rev, ie after sending 100 pulses to it it has made an 8th of a turn, the step-display is 100 pulses, the angle is 45 degrees. 

(NB: actually I'm unsure if that should be 4:1 gearing, not 1:4 ... ie we got the ratio the wrong way round. But I do not know)

urish, 2022-03-31:
Thanks for reporting! Can you check again now?

Looking at [Wikipedia](https://en.wikipedia.org/wiki/Gear_train#Example_2), I think we got it right now. For "4:1", it reads to me: for every 4 revolutions of the stepper motor (=800 steps), there'll be one revolution at the output of the gear system.

M-square, 2022-03-31:
Check on gearRatio. But the step count is still wrong

M-square, 2022-03-31:
Yes, it works as I expect it to work, now.

Now we're "only" missing the other stepper - unipolar (the ubiquous 28BYJ-48)

urish, 2022-03-31:
Any information about the default gear ratio for the 28BYJ-48?

drf5n, 2022-03-31:
The teardown at https://cookierobotics.com/042/ says 32 steps for the motor and 1/64 for the gears, and also: 

> From tooth counts above, gear ratio was found to be 1/64. Gear ratio may vary between manufacturers. For example, Adafruit sells a model that looks identical but rated at 1/16 gear ratio.

so 32 steps on the motor and 64:1 reduction gives 2048 steps/shaft rev.

M-square, 2022-03-31:
My point is that internal gearing is "irrelevant". The only thing that matters (for the Arduino coding) is step/rev, however that is achieved with actual coil/magnets or gear. In RealLife it matters because of torque, speed, backlash etc. but these are not simulated here.

Agree on the 28BYJ-48 has 2048 steps/rev. (Some models have 2047.something ... but it is not specified so it's pot luck what type you get. I've seen other teardowns claim 2038.)

drf5n, 2022-03-31:
Whole steps per rev is a physical property of the motor+gearbox -- Using  drivers with microstepping changes the effective steps/rev.

urish, 2022-03-31:
What should we do first? 28BYJ-48  or a Step/Dir kind of driver?

M-square, 2022-04-01:
Step/Dir as many driverchips do that.

drf5n, 2022-04-01:
> Step/Dir as many driverchips do that.

Many libraries and many programs depend on step/dir as well.

urish, 2022-04-02:
Here's an initial version of the A4988 driver:

https://wokwi.com/projects/327823888123691604

Microstepping implemented, though, the motor itself will only move in half-step increments, due to the limitations of the digital simulation. So if you go with 1/4 steps, the motor will move only once per every 2 step pulses.

To get half-stepping in the example above, connect the MS1 pin to GND.

drf5n, 2022-04-02:
> Here's an initial version of the A4988 driver:
> 
> https://wokwi.com/projects/327823888123691604

Cool
```
    {
      "type": "wokwi-a4988",
      "id": "btn1",
      "top": 85.4,
      "left": 228.27,
      "rotate": 270,
      "attrs": { "color": "green" }
    },
```

drf5n, 2022-04-02:
Here's a UNO + CNC Shield V3 Demo:  https://wokwi.com/projects/327827412674937427

[image]

urish, 2022-04-14:
Documentation for the new Stepper motor driver: https://docs.wokwi.com/parts/wokwi-a4988

Should we keep this issue open for the Unipolar motor? Or can I close it and we'll have the Unipolar as a separate feature request?

M-square, 2022-04-15:
(Terminology correction: The current stepper is a BiPolar (you change polarity, two, bi, to make it turn). The simpler one (28BYJ-48) is Unipolar, you always send the current the same way. https://en.wikipedia.org/wiki/Stepper_motor#Two-phase_stepper_motors. I trip up on that sometimes, too.)

"Both". The original request asks for both types of stepper, but it would prehaps be nicer to "rename" this as **Bipolar** Stepper + Driver and the other as as the Unipolar request, assuming the new one inherits the same Vote-priority. Whatever seems easier for you to adminster.

Personally, I'm happy with that we now have a stepper - thanks, thumbs up, +1 etc

NB: I think there is a problem with the fullwave and halfwave implementation of this Bipolar stepper, and will first do a simple physical implementation to verify/adjust my understanding, and return

urish, 2022-04-15:
> I think there is a problem with the fullwave and halfwave implementation of this Bipolar stepper, and will first do a simple physical implementation to verify/adjust my understanding, and return

Looking forward to your input about this

> The current stepper is a BiPolar

The documentation mentions this: https://docs.wokwi.com/parts/wokwi-stepper-motor
And now I also renamed it in the pop-up where you can add new parts: 
[image]

> Personally, I'm happy with that we now have a stepper - thanks, thumbs up, +1 etc

That's why I asked if we still want to keep this issue open - there are plenty of other feature requests waiting for love. If whoever voted for this features cares about the unipolar motor and feels it's important to have it, I'll do it and then close then issue. Otherwise, I will just move on to working on other stuff (e.g. the ability to move wires, project collections, interactive debugger, breadboard support, etc.)

M-square, 2022-04-15:
I suppose I am saying: close this one. It is just a minor (software)driver issue of distinction between them, not a fundamental missing item like it was before with no stepper. (Steppers and servos are SO different) But others can chime in here, too.

I'll buy you a beer (= buy a vote) for the unipolar at some time - when I open that feature request, just to get it into the queue.

closed 2022-04-18 by urish