wokwi/wokwi-features #193: Arduino Mega - Pin 3: Not generating the interrupts

ethanengelander, 2021-09-25:
Pin 3 does not generate the hardware interrupts or causes an internal error [assumption].

I have also noticed that all interrupt pins behave differently to the other interrupt pins. 

There also might be some issues with pin 18, unless that one is the most accurate [I don't understand the exact behaviour to know if this pin is bugged, behaving correctly, or if this pin is the only one that works].

I have written up a sketch for all interrupt pins to be connected to their own individual button as a hardware interrupt on the mega:
https://wokwi.com/arduino/projects/310687432027669056

urish, 2021-09-25:
Seems like the issue is that the pins are floating inputs. Does setting the pin mode to `INPUT_PULLUP` solve the issues?

urish, 2022-03-27:
Closing due to lack to activity

closed 2022-03-27 by urish