wokwi/wokwi-features #195: Accessibility and responsiveness on small screens (smartphone or tablets)

djedu28, 2021-09-27:
I need to use an Arduino simulator for robotics lessons. I liked Wokwi, mainly because it's open source.

Whoever I try to bring to the world of robotics still cannot afford to buy a kit, so the idea of ​​simulating, even worse, doesn't even have a computer at home, only smartphones.

Hence my need to use Wokwi on small screens. 

On the project page I need a button to activate/deactivate the editing of the code, so that the device keyboard does not appear in an accidental click or when walking in the code.

I also need a button to activate the code tab's full screen, just as there is a button to activate the simulation's full screen.

Having an option to order the "code" and "simulation" tabs vertically instead of horizontally will help a lot for mobile operation.

These are the main features.

Other needs are:
a. X button to confirm deleting a component (clicks by the phone end up deleting a component from the simulation).

b. Zoom button: zoom +, zoom - and zoom 100%. can be inserted in the upper corner of the simulation.

Forgive the bad English, I only speak Portuguese, I'm using the Google translator.

*I can help you develop some of these features if you like. I'm junior in web development with react.*

Obs. I love your work, Wokwi and all other repositories and components are amazing! Congratulations and thank you for allowing us to study the source code and for the richness of detail in the documentation.

urish, 2021-09-27:
Nice to meet you Edurado and thank you for the kind words! 

I feel your pain. Not too long ago I started working on a mobile-optimized version. You can see a preview version here:

https://new-mobile.preview.wokwi.com/arduino/new

As for the other needs: 
a. Button for confirming the deletion of components - see #188 
b. Zoom button: see #127 

The zoom button is currently almost [on top of our public feature list](https://wokwi.com/features) so it's very likely to happen soon. 

Have you seen our discord chat at https://wokwi.com/discord? There's a #user-interface channel where we discuss this kind of stuff

urish, 2021-12-22:
Closing as we released the mobile optimized version, and the remaining concerns are addressed by other feature requests (see my previous comment)

closed 2021-12-22 by urish