wokwi/wokwi-features #196: Franzininho WiFi Board (ESP32-S2)

arcostasi, 2021-09-28:
Franzininho WiFi Board is a development board to evaluate ESP32-S2 Modules (Wroom and Wrover) and develop the new generation of Franzininho Boards. The Franzininho project was created to develop skills in people in the areas of electronics and programming, through activities in the DIY format and in conjunction with maker culture in Brazil.

Project: https://github.com/Franzininho/Franzininho-WIFI

The project is an open-source hardware and is available on CERN open hardware license.

The Franzininho WiFi board is certified in OSHWA: [UID BR000006](https://certification.oshwa.org/br000006.html)

urish, 2021-09-29:
Great to see this racing to the top of the [feature list](https://wokwi.com/features)!

Preparing the data for the board design:
- [Pinout](https://www.embarcados.com.br/wp-content/uploads/2021/09/pinagem-franzininho-wifi.png)
- [Board dimensions](https://github.com/Franzininho/Franzininho-WIFI/blob/main/Franzininho%20WIFI/pdf/franzininhoWIFI-pcb.pdf)

Is there any publicly available SVG drawing of the board?

urish, 2021-10-05:
Initial version of the board is now available in [wokwi/wokwi-boards](https://github.com/wokwi/wokwi-boards/tree/master/boards/franzininho-wifi). 

You can go ahead and load it in the simulator by following the [instructions for loading a custom board](https://github.com/wokwi/wokwi-boards#testing-a-custom-board). 

You'll be able to preview the board and draw connections, but the simulation does not work yet:
1. The build will fail due to espressif/arduino-esp32#5735
2. The ESP32-S2 chip simulation is still WiP

urish, 2021-10-08:
> Thanks Uri, any results with the armel toolchain mentioned in espressif/arduino-esp32#5735 (comment)

Yes, positive results! I just added a comment there

urish, 2021-10-09:
Some more updates: working to get the ESP32-S2 simulator to a usable state, then integrate it with Wokwi.

So far, was able to get MicroPython 1.17 prompt:

```
$ npx ts-node demo/micropython

MicroPython v1.17-74-gd42cba0d2-dirty on 2021-10-09; ESP32S2 module with ESP32S2
Type "help()" for more information.
>>> print("Hello, world!", 3**6)
Hello, world! 729
>>> import time
>>> time.ticks_ms()
945648215
>>> 
```

So this is a good sign. It means that the following peripherals are working correctly:
- CPU core
- UART
- Flash access (otherwise MicroPython fails to start with FS initialization errors)
- System timer (partially - we shouldn't get a ticks_ms() value this large when we just started)

There's still much work to do before it'll be usable:
- USB (as Franzininho WiFi uses USB over UART)
- GPIO (so we can eventually see the LED blinking!)
- Fix the timers (so delay() works reliably)

And probably a bunch of other things we'll discover along the way :-)

urish, 2021-10-09:
Excellent! CircuitPython will probably be another stage, first let's get the Arduino stuff to work!

urish, 2021-10-17:
Video showing CircuitPython work-in-progress: https://twitter.com/WokwiMakes/status/1449804926478168068

urish, 2021-10-31:
Update: you can now run the [Franzininho WiFi CircuitPython Demo Project](https://wokwi.com/arduino/projects/313606939786347074) out of the box, without any special instructions.

I hope the Arduino Core issue gets fixed soon, and so we'll also have a set of demos with Arduino.

arcostasi, 2021-11-02:
Very good, doing some tests with CircuitPython, a question, is time.sleep already implemented?

This is the code I'm testing:

```python
""" Hello, World! (Blink LED)"""
import board
import time
from digitalio import DigitalInOut, Direction

# Setting the LED pin, the LED I configured was pin 4 = IO4
led = DigitalInOut(board.IO4)
led.direction = Direction.OUTPUT

#loop infinito - executando sempre
while True:    
    led.value = True
    time.sleep(0.5)
    led.value = False
    time.sleep(0.5)
```

Diagram link: https://wokwi.com/arduino/projects/314165931402068544

FBSeletronica, 2021-11-02:
> Very good, doing some tests with CircuitPython, a question, is time.sleep already implemented?
> 
> This is the code I'm testing:
> 
> ```python
> """ Hello, World! (Blink LED)"""
> import board
> import time
> from digitalio import DigitalInOut, Direction
> 
> # Setting the LED pin, the LED I configured was pin 4 = IO4
> led = DigitalInOut(board.IO4)
> led.direction = Direction.OUTPUT
> 
> # infinite loop - always running
> while True:    
>     led.value = True
>     time.sleep(0.5)
>     led.value = False
>     time.sleep(0.5)
> ```
> 
> Diagram link: https://wokwi.com/arduino/projects/314165931402068544

I had the same problem here: 
[image]

urish, 2021-11-02:
Yes, it seems like the internal timer is going way too fast, so the sleep finishes immediately. I'll try to figure out why this happens

urish, 2021-11-04:
Sleep fixed!

urish, 2021-11-04:
Progress update: Filesystem works!

You can try it here: https://wokwi.com/arduino/projects/313606939786347074

Note that the code still doesn't run automatically (CP starts in Safe mode, I still have to figure out how to avoid this), so you have to manually load it from the REPL:

```python
import main
```

urish, 2021-11-10:
espressif/arduino-esp32#5735 fixed! Meaning you can now enjoy:

- [RGB Blink on Franziniho WiFi using the Arduino Core](https://wokwi.com/arduino/projects/314916550651937346)

There's still some issue with the USB Serial output on Arduino Core, I have to debug it and figure out why it doesn't print any output. But other than that, it's pretty usable :)

arcostasi, 2021-11-11:
Thanks Uri, good to hear that espressif fixed this! everyone is to be congratulated 😉

urish, 2021-11-23:
USB Serial on Arduino core works! Example:

https://wokwi.com/arduino/projects/316092312585241154

So the Arduino core is pretty much complete. Let's see what people build with it. Over the next few days, I'll add the Franzininho WiFi to the "new project" menu, documentation, etc. so it'll become generally available.

My next action item here is to make CircuitPython load the user code when the simulation starts.

urish, 2021-11-23:
Forgot to say - also added an automated test case for Franzininiho WiFi simulation: https://github.com/wokwi/wokwi-tests/tree/master/franzininho-wifi/usb-serial

arcostasi, 2021-11-23:
Thanks Uri, I'll share it in the community and try it out!

urish, 2022-02-04:
Some updates: I implemented a bunch of performance speedups (similar to the ones I did for the ESP32), and now the ESP32-S2 simulation runs about 4 times faster.

Getting the WiFi to work is still a big challenge - I'll keep update here as I slowly make progress.

arcostasi, 2022-02-05:
> Some updates: I implemented a bunch of performance speedups (similar to the ones I did for the ESP32), and now the ESP32-S2 simulation runs about 4 times faster.

That's great Uri, it's already a huge improvement! In my tests I've already noticed the improvement in speed ;)

> Getting the WiFi to work is still a big challenge - I'll keep update here as I slowly make progress.

All right, but that's already a big step forward! I'll do more tests! Thanks

urish, 2022-02-08:
Good news: the WiFi is working!

Examples:
- [NTP Time](https://wokwi.com/arduino/projects/323043284024820308)
- [HTTP Server](https://wokwi.com/arduino/projects/323043611101889108) - Requires the [Wokwi IoT Gateway](https://docs.wokwi.com/guides/esp32-wifi#the-private-gateway)

Note: TLS support isn't there yet. We'll need to implement the crypto hardware accelerators - SHA, AES, RSA and possibly a few more before TLS/HTTPS will be functional. Similar to what we did in #267.

arcostasi, 2022-02-08:
What good news! I did some experiments and everything worked just fine! The performance is also excellent!

Here is my test example:

- [Weather HTTP Server](https://wokwi.com/arduino/projects/323063403217158738) - Requires the [Wokwi IoT Gateway](https://docs.wokwi.com/guides/esp32-wifi#the-private-gateway)

Thanks Uri

urish, 2022-02-09:
Very cool!

[image]

Did you have any problems getting the DHT sensor to work?

FBSeletronica, 2022-02-09:
Amazing Uri! I will test and make some videos. An article on the "Embarcados" website is also worth sharing with the Brazilian community. This is a piece of good news! Thank you.

arcostasi, 2022-02-12:
> Did you have any problems getting the DHT sensor to work?

Yes Uri, it is returning this value:
Humidity: nan
Temperature: nan

with esp32 nodemcu it's ok

urish, 2022-02-12:
> Yes Uri, it is returning this value:

I'm not sure why - but if you move it to pin 11 it seems to work correctly

urish, 2022-02-16:
More news: I2C is now working on the Franzininho WiFi!

Sample project: 
* [Franzininho WiFi + LCD1602 over I2C Clock](https://wokwi.com/arduino/projects/323796775459619410)

arcostasi, 2022-02-18:
That's great! I will share it with the community! Thanks

urish, 2022-02-20:
Progress update: HTTPS is now working on Arduino Core!

I made a quick example that fetches jokes from an online REST API and displays them on an LCD screen:

https://wokwi.com/arduino/projects/324148356078109268

I've finished the AES/SHA accelerators. Once I finish the RSA accelerator, I hope we'll also get HTTPS on CircuitPython.

urish, 2022-02-26:
@arcostasi yes - but you need to adjust the connection in diagram.json to read:

```
    [ "esp:TX", "$serialMonitor:RX", "", [] ],
    [ "esp:RX", "$serialMonitor:TX", "", [] ],
```

(instead of TX0 / RX0)

arcostasi, 2022-02-26:
Thanks Uri, printf() is working perfectly! :)

urish, 2022-02-26:
Excellent!

What do you think is still missing before we can release the Franzininho WiFi board?

urish, 2022-02-26:
Thank you! 

From my end: need to create a documentation page, add it to the list of "New project" options, create some basic examples (perhaps we can use the examples that you created?)

Nice-to-have: Finish CircuitPython support (so it won't start in Safe mode); MicroPython support

arcostasi, 2022-02-26:
> From my end: need to create a documentation page, add it to the list of "New project" options, create some basic examples (perhaps we can use the examples that you created?)

Feel free to use my examples! I will create more examples during my tests.

> Nice-to-have: Finish CircuitPython support (so it won't start in Safe mode); MicroPython support

Yes, I believe CircuitPython will be used more by the Franzinho community.

Thanks Uri

arcostasi, 2022-02-26:
@urish, I'm testing the inputs, in your example of the piano with Franzininho WiFi, when I release a key and press another the input doesn't respond, I need to press it again to hear the buzzer.

- [Mini piano example](https://wokwi.com/arduino/projects/324676273947279956)

urish, 2022-02-26:
Thanks @arcostasi!

Can you connect the logic analyzer and check if the data is fed in (but not played but the speaker) or if it skips feeding in the data?

On a different note - worked on CircuitPython today. It now runs your code automatically:

https://wokwi.com/arduino/projects/313606939786347074

However, there's one annoying thing: you don't see all the output immediately, as it takes a few microseconds for the USB to connect. This means that if you have a syntax error in your code, you'll need to press "Ctrl+D" to see it.

arcostasi, 2022-02-27:
Hi @urish,  apparently with the logic analyzer it's normal! I pressed 3 times alternately between the red and orange button and I did not hear the speaker, when I pressed 2 times on the same button without alternating I can hear the speaker.
[image]

urish, 2022-02-27:
Thanks! Can you please also connect the speaker pin to the logic analyzer?

arcostasi, 2022-02-27:
> However, there's one annoying thing: you don't see all the output immediately, as it takes a few microseconds for the USB to connect. This means that if you have a syntax error in your code, you'll need to press "Ctrl+D" to see it.

That is great! as soon as I have some free time I'll have fun with CircuitPython.

arcostasi, 2022-02-27:
> Thanks! Can you please also connect the speaker pin to the logic analyzer?

A square wave appeared in the speaker channel (D2), but the audio did not play in my headphones.

[image]

urish, 2022-02-27:
Thanks! So I suspect an issue with the buzzer. Does the frequency of the square wave look correct? (you can ask PulseView to calculate the frequency by holding Shift and selecting the region between two adjacent peaks (low-to-high transitions).

Also, do you observe any difference when the buzzer is set to [accurate mode](https://docs.wokwi.com/parts/wokwi-buzzer#operation-modes)?

arcostasi, 2022-02-27:
In accurate mode I can hear the sound with noises, even switching between the buttons.

[image]

urish, 2022-02-28:
Thanks @arcostasi. I think I found the issue with the smooth mode - it looked at edge-to-edge time instead of peak-to-peak time to determine the frequency. The ESP32 uses a non-50% duty cycle, so it confused the smooth mode algorithm:

[image]

You can see that the frequency is right, but the duty cycle (% of the time the signal is high in each cycle) is not 50%. This is different from how Arduino does PWM. 

I have updated the algorithm, can you give it another go please?

arcostasi, 2022-02-28:
Hi @urish, it worked perfectly now! Thank you

arcostasi, 2022-03-01:
@urish, I did another test with Franzinho WiFi, was the analog reading implemented? Thanks

Here is an example: [RGB LED with Slide Potentiometer](https://wokwi.com/arduino/projects/324682033130373716)

urish, 2022-03-01:
> was the analog reading implemented?

As far as I can tell not yet

arcostasi, 2022-03-01:
Ok, no problem! Thank you

urish, 2022-03-01:
Update: analogRead (ADC) implemented!

arcostasi, 2022-03-01:
Thanks @urish, worked perfectly!

arcostasi, 2022-03-24:
Here's an example of using a [servo with Franzininho WiFi and ESP-IDF](https://wokwi.com/projects/326685017518375507)

urish, 2022-04-15:
Documentation for the Franzininho WiFi board: https://docs.wokwi.com/parts/board-franzininho-wifi

urish, 2022-04-16:
The last missing feature is now ready: SPI support. Here's a quick demo:

https://wokwi.com/projects/329013233501340242

So now we have feature parity with the ESP32 and can declare the Franzininho WiFi simulation ready. Thanks for everyone who supported this feature :-)

closed 2022-04-16 by urish

urish, 2022-04-18:
Franziniho WiFi now has a "New project" page: https://wokwi.com/projects/new/franzininho-wifi