wokwi/wokwi-features #197: We need a FT6206 simulated for the cap touch screen.

leftCoast, 2021-09-30:
The Adafruit 1947 incorporates A TFT, capacitive touch and a SD card reader. Well, we have all the bits except the FT6206 touch chip. So, Lets add one of those in and we can really have some fun!

-jim lee

urish, 2021-09-30:
FT6206 datasheet: https://cdn-shop.adafruit.com/datasheets/FT6x06+Datasheet_V0.1_Preliminary_20120723.pdf

urish, 2021-10-02:
Update: there's a working prototype. Would you like a preview link to play with it or wait a day or two until it's ready?

urish, 2021-10-05:
FT6206 demo: https://wokwi.com/arduino/projects/311598148845830720

Note that you need to add a custom board for this to work, see the instructions at the top of the file.

leftCoast, 2021-10-05:
Uri, I may not be able to test this for a bit. At sea for a couple more
weeks. If I get near cell coverage I can at least do eMail (Today).

-jim

leftCoast, 2021-10-06:
> No worries, whenever you are ready!
>
>
> What are you doing at sea?

On a Holland America cruise ship going from Seattle to Florida via the
Panama canal. 3 weeks.

At this moment we are sitting dead in the water between Catalina island
(Off of Los Angeles) and San Diego. Its night time and wonder if the motor
broke? Oh well.. We'll see.

-jim lee

urish, 2021-10-06:
Captain Jim Lee!

urish, 2021-10-31:
Ahoy Captain! 

The Capacitive Touch Screen is ready for your command :-)

https://wokwi.com/arduino/projects/311598148845830720

closed 2021-10-31 by urish

leftCoast, 2021-10-31:
Great! Give me some time I have some code to try on it.

urish, 2021-10-31:
Lovely! Take as much time as you need :)

Koepel, 2021-11-03:
There is a offset between the mouse and the pixels with Firefox 93.0 in Ubuntu linux, using the project mentioned above: https://wokwi.com/arduino/projects/311598148845830720
It does not happen with other browsers, it does not even happen with Firefox in an other linux distro (Manjaro) !

https://user-images.githubusercontent.com/12096541/140102285-4ded8685-d6c6-4ff4-9e7a-8900e2e17d67.mp4

I don't know why my specific setup causes this.

Koepel, 2021-11-03:
With a Guest session I tried Firefox with its default settings in Ubuntu linux and there was no offset problem.
That means that my specific settings and add-on causes a offset for this display in Wokwi (and no where else).

Koepel, 2021-11-03:
The add-on Font Fingerprint Defender is the cause of the offset: https://addons.mozilla.org/nl/firefox/addon/font-fingerprint-defender/

urish, 2021-11-03:
Installing this extension reproduces the issue for me on Windows, but to a lesser extent. I guess the fingerprinting protection somehow interferes with some of the mouse pointer calculations.