wokwi/wokwi-features #198: seven segment display doesnt appear good when run at low speed

matititam, 2021-10-04:
Due to my PC limitation, the Seven segment display project (ESP32 Arduino core simulation) looks not so normal
[image]

Project link: https://wokwi.com/arduino/projects/305567166302782017

matititam, 2021-10-15:
The fix is working! The display is now very smooth and natural

tumypv, 2022-02-11:
7 segment with Arduino still dosn't work properly on high speed.
https://wokwi.com/arduino/projects/323324789238268498

urish, 2022-02-12:
> 7 segment with Arduino still dosn't work properly on high speed.

@tumypv thanks for reporting! Mind opening a new issue for this?