wokwi/wokwi-features #203: Resistor divider simulation

urish, 2021-10-09:
Resistor dividers are useful for connecting multiple buttons to a single analog input pin, as [described here](http://www.ignorantofthings.com/2018/07/the-perfect-multi-button-input-resistor.html). This is particularly useful for MCUs with limited number of IO pins, such as the ATtiny85.

Currently, Wokwi includes a very basic analog simulation: You can connect a single part (e.g. [a potentiometer](https://docs.wokwi.com/parts/wokwi-potentiometer) or an [analog temperature sensor](https://docs.wokwi.com/parts/wokwi-ntc-temperature-sensor)) to one or more analog input pins. However, if you add any [resistors](https://docs.wokwi.com/parts/wokwi-resistor) to your analog circuit, the simulator will ignore them.

Let's simulate them!

MykolaKyrylenko, 2021-10-10:
Great idea. I agree.
I am not great on GiHub, but is there a formal way to vote?

urish, 2021-10-10:
Thank you! The way the vote is on the [features page](https://wokwi.com/features), using VotePowers. You can vote as many times as you want to feature. I'm still working on a more detailed FAQ that explains how this works, but here is a [quick explainer](https://docs.google.com/document/d/10_MW0M7x9elBbBl6ZaXMg08i8AWQHS2GqJdXxUzNwXQ/edit#).

JCalvi, 2022-07-13:
This would be great.

In the interim would it be possible to make the potentiometer have hot keys for values so we can simulate the same thing?
At the moment Home = 0, End = 1023. It would be great to add other keys and mappings for that device, eg 1 = 100, 2 = 200 etc so we can simulate buttons on a voltage divider by just connecting the potentiometer and pressing one of the hot keys?

urish, 2022-07-20:
> In the interim would it be possible to make the potentiometer have hot keys for values so we can simulate the same thing?

Sounds like a good suggestion. There's also page-up/page-down that jump up/down, but not to a specific point. If anyone is interested in hacking on this, you can add a new keyboard event listener [here](https://github.com/wokwi/wokwi-elements/blob/0688efa73dfdeccf805ae4e296e1514ddb483c15/src/slide-potentiometer-element.ts#L57-L69), and then check if the key is 1, 2, 3, etc. and modify the value accordingly.

urish, 2022-08-12:
While waiting for this, there's some workaround: you can achieve a similar functionality by implementing a custom chip (see #301 for details). Here's one example showing how to use a custom chip to simulate multiple analog buttons connected to a single analog input:

https://wokwi.com/projects/339700374803841619

JCalvi, 2022-12-12:
The custom chip works well BTW. Thanks for the suggestion.

kubark42, 2023-03-03:
Oddball suggestion, but instead of implementing further simulation in the backend, maybe borrow the simulator from https://github.com/SEVA77/circuitjs1?

P.S. This is an awesome project. I'm so glad I found it.

urish, 2023-08-01:
Thanks for the suggestion @kubark42! 

The funny thing about "circuitjs1" is that it's actually implemented in Java and compiled to JavaScript.

In any case, we already have a solution for the analog simulation part of this feature. The main challenge is how to integrate it into the existing simulation framework while keeping the performance high and not breaking existing projects which rely on the current behavior of resistors.

Right now, this feature is #8 in the [requested feature list](https://wokwi.com/features), let's see how long until it makes it to the top :-)

martinius96, 2023-09-18:
2 years later, still not implemented

urish, 2023-09-18:
1.5 months later, and now it's number 6 on [the list](https://wokwi.com/features)

kubark42, 2023-09-18:
Prioritization should happen for _important_ things and _urgent_ things. It isn't a great optic when it takes 2 years to do something as simple as a resistor divider. Maybe it's a heck of a lot of work on the backend, and that's why it isn't happening. But if prioritization comes down only to votes, then low-hanging fruit never gets prioritized. Typos and documentation errors will endure effectively indefinitely if only the most highly voted features get implemented.

Looking at the prio list, some things are jaw-droppingly difficult, such as an offline simulator. Le'ts hope that resistor dividers are easy and can find their way into the queue before the devs tackle offline support and its almost infinite number of computer configurations.

urish, 2023-09-19:
I'm wondering - how much work time do you estimate the resistor divider would take?

martinius96, 2023-09-19:
Nothing against you, but this functionality of voltage dividers is already in the worst simulator TinkerCAD...
That does not have other important features for which I switched to Wokwi.

Dont know how simulation in Wokwi works, if it is not like "circuit" simulation, but more cpu only simulation with peripherals.

jxmked, 2025-03-17:
Is this implemented? Couple hours and i got no success until i found this. Painfull. Btw, this is good...

urish, 2025-03-18:
Not yet

djedu28, 2025-03-27:
> Is this implemented? Couple hours and i got no success until i found this. Painfull. Btw, this is good...

@jxmked You can simulate this behavior with a *Custom Chip*