wokwi/wokwi-features #204: Offline version of the simulator

urish, 2021-10-09:
A version of the simulator you can use without an internet connection. Either by making the online version a PWA (progressive web app) that is available offline, or by providing a downloadable version of the application. In any case, this will require installing an Arduino compiler on the local machine (in order to compile the code).

Note that the Offline version is very likely to be a paid feature. Anyone who votes for this feature (before the first version is released) will get this feature, including one year of updates. [Wokwi club members](https://wokwi.com/club) will also get access to the offline version.

urish, 2021-10-21:
> A version that can be integrated into an LMS (e.g. Moodle) will be great.

You mean, a way to embed simulation projects into Moodle? Or do you have in mind a different kind of integration?

MykolaKyrylenko, 2021-10-21:
What what I have seen, there are several ways to add to Moodle.
- With Falstad, I managed to compile the code and upload the resultant website as a file onto Moodle. Details here: https://help.lafayette.edu/importing-a-website-into-moodle/
- I have not done this, but another option could be H5P, which is HTML5 content

Note: the Falstad implementation includes AVR8js which works well on our Moodle

gauravacad, 2023-08-23:
cannot we use limited data for IoT gateway http server. always having club feature is not possible

sasdallas, 2023-12-05:
@gauravacad Assuming there is no solution, is there a way to mark this as unaddable or so? As disappointing as it is, if it actually cannot happen, most members shouldn't waste their votepowers on it.

urish, 2023-12-10:
> As disappointing as it is, if it actually cannot happen, most members shouldn't waste their votepowers on it.

Are you asking about Offline version of the simulator? It's in development. It'll be available as a feature in Wokwi for VS Code, and perhaps also as a totally standalone version after, depending on demand.

urish, 2023-12-27:
Wokwi for VS Code now has initial support for offline operation, as a paid add-on.  You can purchase it when you activate your Wokwi for VS Code license, from this page: https://wokwi.com/license.

I would love to hear your feedback - does it answer your needs? Is anything missing?

sasdallas, 2024-01-10:
@urish While I understand the fact that bills need to be paid, $97 a year seems quite pricy, especially for those with big projects who just want to compile it faster.
I haven't tried it out yet but maybe adding a monthly payment option or a free trial might be helpful! This definitely seems like something bigger than Wokwi Club, so it does make sense to have it as an addon.

Thank you for your work!

urish, 2024-01-14:
Thanks for the feedback @sasdallas! I'll look into adding a monthly payment option.

Timmmm, 2024-03-12:
I think it would be worth clarifying somewhere whether Wokwi sends code to your servers or not if you *don't* buy the offline feature. Your privacy policy doesn't say anything about this - is the online requirement for the free version of the VSCode extension just a "check we're online" type thing or is it actually sending stuff to servers?

Timmmm, 2024-03-16:
Very helpful thanks! Though I'm not sure you can "leave questions open" in your privacy policy. If you don't say you are using some data then you aren't allowed to use it. However I think you can just update the privacy policy if/when you want to change it.

selmen2004, 2024-03-17:
@urish I have complied an offline version using only ESP32 / micropython for educational use only ( we teach using wokwi in our secondary schools in Tunisia and some schools have internet issues )
is this ok  ?

urish, 2024-03-18:
> @urish I have complied an offline version using only ESP32 / micropython for educational use only ( we teach using wokwi in our secondary schools in Tunisia and some schools have internet issues )

Would love to see what you compiled - how can I check it out?

selmen2004, 2024-03-18:
> > @urish I have complied an offline version using only ESP32 / micropython for educational use only ( we teach using wokwi in our secondary schools in Tunisia and some schools have internet issues )
> 
> Would love to see what you compiled - how can I check it out?

Sure , here  : https://selmen.tn/2023/04/28/wokwi-esp32-micropython-simulator-offline/

saturnquack, 2024-04-04:
Any update on an offline version of the non-vs code one? From what I'm reading, its a maybe?

urish, 2024-04-07:
> Any update on an offline version of the non-vs code one? From what I'm reading, its a maybe?

Nothing official at the moment :)

kyvaith, 2024-10-14:
Does the free version of the Wokwi extension for VSCode do the simulation/emulation in the cloud, while the offline version does it locally? I'm trying to find a way to speed up the emulation, because now trying to emulate LVGL stuff gives me 1-5 FPS compared to 60 FPS on the real ESP32.

urish, 2024-10-14:
> I'm trying to find a way to speed up the emulation, because now trying to emulate LVGL stuff gives me 1-5 FPS compared to 60 FPS on the real ESP32

Which display are you using and which specific ESP32 chip? Speeding up LVGL simulation is a real challenge.

kyvaith, 2024-10-14:
I'm using the only avaliable touchscreen display in Wokwi. The Chip is ESP32-S3.

urish, 2024-10-14:
Yeah, the S3 is especially challenging, since we have to simulate two cores. If you can recreate a minimal reproduction of this and open a new issue, we can take a look if there are any quick wins (but I wouldn't expect more than x2-x3 speed up compared to the current simulation speed, without a significant amount of optimization work)

kyvaith, 2024-10-14:
I'm using ESPhome in VSCode, so probably no good example to bo pasted in the online version of Wokwi. I can provide you compiled firmware or you can compile it from my code https://github.com/kyvaith/atmosfera-echo-hub

urish, 2024-10-14:
Thanks, can you please open a separate issue to track this? also, please explain how to compile the code in the repo (I see some files there, but there are no sources, as I assume there's some build process with esphome)

kyvaith, 2024-10-14:
Ok, but the main question is if the Wokwi emulates ESP32 localy or in cloud. If it's localy maybe I can just use machine with more powerfull CPU? Regarding the compilation proces - just use the instructions from my repo.

urish, 2024-10-14:
Locally

brentru, 2024-11-19:
@urish  Could you please clarify if the Offline version of the simulator is only for VSCode, or can we use it for running Wokwi-cli test scenarios offline (instanciated from a terminal)?

urish, 2024-11-19:
@brentru there's an offline version of the wokwi-cli server that can run in docker, so it can be used offline. Would love to hear more about your use case

sasdallas, 2024-12-27:
> Payed? Greedy as shit of something to do tbh

Not at all. It is not easy to create a simulator that can handle this level of firmware emulation and having devices act as they should in simulation. Emulating an Arduino is hard work, emulating an arduino that has I/O busses and communicates with other peripherals is even more work.

While I agree with you that the current price is _way_ out of proportion (at least when I checked), being paid isn't a greedy move.

salvalie, 2025-04-29:
@MicrocadeGuy It's a proprietary software, you are in no position to discuss their business model. Just dont buy it. Is it too hard to understand? I am looking into foss alternatives, but those aren't even close to this software.