wokwi/wokwi-features #205: Interactive Debugger (Breakpoints, Stepping, Watches)

urish, 2021-10-12:
Add a built-in visual debugger to Wokwi. Wokwi currently has a [powerful GDB integration](https://docs.wokwi.com/gdb-debugging). However, GDB has a steep learning curve, and beginners often have hard time working with it.

An integrated visual debugger can solve this. Ideally, we'd have the following features:
- Line breakpoints
- Stepping (run the code line-by-line)
- Watches (inspect the values of local/global variables while the program is stopped in the debugger)

lemio, 2022-02-20:
This would be great for educational purposes, showing how the hardware is influenced by stepping through the lines of code could really help beginners to conceptualise time, states and conditions. I've been creating some GIF's in the past to do this.

[image]

Using watches in combination with seeing the (simulated) hardware could also highly benefit the understanding of mapping and control.

[image]

urish, 2022-02-22:
These GIFs look great, thanks for sharing them :)

urish, 2022-04-19:
The program can be compiled in two ways: release mode, and debug mode. In debug mode, the compiler skips some of the optimizations, to make it easier to step through the code in the debugger. When you run Web GDB in Wokwi, then Wokwi rebuilds the project in debug mode. 

So I'm guessing that this is the reason you are seeing different addresses: when you built it locally, you got the release mode HEX, but when running inside Wokwi with WEB GDB, it uses the debug mode HEX. Does it make sense?

ricardojlrufino, 2022-04-19:
Got it, it makes sense. I was using the HEX downloaded from the IDE, this is the "RELEASE" version correct?
[image]

urish, 2022-04-19:
Correct

urish, 2022-04-19:
When we build the Debug version on Wokwi, we use the Arduino CLI and add the following flag to disable compiler optimizations: `--build-property build.extra_flags=-Og`

ricardojlrufino, 2022-04-19:
I did this test, the addresses change but still differing...
[image]

I'm changing platform.txt
[image]

urish, 2022-04-19:
Are you using the latest version of the Arduino AVR core? same library versions (in case you are using any libraries)?

ricardojlrufino, 2022-04-19:
arduino-1.8.16
avr-gcc/7.3.0-atmel3.6.1-arduino7

```
// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  asm("BREAK;");
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}
```

ricardojlrufino, 2022-04-19:
Compile log
```
../arduino-1.8.16/arduino-builder -dump-prefs -logger=machine -hardware ../arduino-1.8.16/hardware -hardware ../arduino-1.8.16/portable/packages -tools ../arduino-1.8.16/tools-builder -tools ../arduino-1.8.16/hardware/tools/avr -tools ../arduino-1.8.16/portable/packages -built-in-libraries ../arduino-1.8.16/libraries -libraries ../arduino-1.8.16/portable/sketchbook/libraries -fqbn=websim:avr:websim-uno -ide-version=10816 -build-path /tmp/arduino_build_987778 -warnings=none -build-cache /tmp/arduino_cache_570185 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.webuploader.path=../arduino-1.8.16/portable/packages/websim/tools/webuploader/1.0 -prefs=runtime.tools.webuploader-1.0.path=../arduino-1.8.16/portable/packages/websim/tools/webuploader/1.0 -prefs=runtime.tools.avr-gcc.path=../arduino-1.8.16/portable/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-4.8.1-arduino5.path=../arduino-1.8.16/portable/packages/arduino/tools/avr-gcc/4.8.1-arduino5 -prefs=runtime.tools.avrdude.path=../arduino-1.8.16/portable/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=../arduino-1.8.16/portable/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.arduinoOTA.path=../arduino-1.8.16/portable/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=../arduino-1.8.16/portable/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=../arduino-1.8.16/portable/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -verbose /tmp/arduino_modified_sketch_752898/Blink.ino
../arduino-1.8.16/arduino-builder -compile -logger=machine -hardware ../arduino-1.8.16/hardware -hardware ../arduino-1.8.16/portable/packages -tools ../arduino-1.8.16/tools-builder -tools ../arduino-1.8.16/hardware/tools/avr -tools ../arduino-1.8.16/portable/packages -built-in-libraries ../arduino-1.8.16/libraries -libraries ../arduino-1.8.16/portable/sketchbook/libraries -fqbn=websim:avr:websim-uno -ide-version=10816 -build-path /tmp/arduino_build_987778 -warnings=none -build-cache /tmp/arduino_cache_570185 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.webuploader.path=../arduino-1.8.16/portable/packages/websim/tools/webuploader/1.0 -prefs=runtime.tools.webuploader-1.0.path=../arduino-1.8.16/portable/packages/websim/tools/webuploader/1.0 -prefs=runtime.tools.avr-gcc.path=../arduino-1.8.16/portable/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-4.8.1-arduino5.path=../arduino-1.8.16/portable/packages/arduino/tools/avr-gcc/4.8.1-arduino5 -prefs=runtime.tools.avrdude.path=../arduino-1.8.16/portable/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=../arduino-1.8.16/portable/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.arduinoOTA.path=../arduino-1.8.16/portable/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=../arduino-1.8.16/portable/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=../arduino-1.8.16/portable/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -verbose /tmp/arduino_modified_sketch_752898/Blink.ino
Using board 'websim-uno' from platform in folder: ../arduino-1.8.16/portable/packages/websim/hardware/avr/1.0.0
Using core 'arduino' from platform in folder: ../arduino-1.8.16/portable/packages/arduino/hardware/avr/1.8.5
Detecting libraries used...
../arduino-1.8.16/portable/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10816 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -Og -I../arduino-1.8.16/portable/packages/arduino/hardware/avr/1.8.5/cores/arduino -I../arduino-1.8.16/portable/packages/websim/hardware/avr/1.0.0/variants/standard /tmp/arduino_build_987778/sketch/Blink.ino.cpp -o /dev/null
Generating function prototypes...
../arduino-1.8.16/portable/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10816 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -Og -I../arduino-1.8.16/portable/packages/arduino/hardware/avr/1.8.5/cores/arduino -I../arduino-1.8.16/portable/packages/websim/hardware/avr/1.0.0/variants/standard /tmp/arduino_build_987778/sketch/Blink.ino.cpp -o /tmp/arduino_build_987778/preproc/ctags_target_for_gcc_minus_e.cpp
../arduino-1.8.16/tools-builder/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /tmp/arduino_build_987778/preproc/ctags_target_for_gcc_minus_e.cpp
Compilando sketch...
../arduino-1.8.16/portable/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10816 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -Og -I../arduino-1.8.16/portable/packages/arduino/hardware/avr/1.8.5/cores/arduino -I../arduino-1.8.16/portable/packages/websim/hardware/avr/1.0.0/variants [... cut ...]
../arduino-1.8.16/portable/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-objcopy -O ihex -R .eeprom /tmp/arduino_build_987778/Blink.ino.elf /tmp/arduino_build_987778/Blink.ino.hex
../arduino-1.8.16/portable/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-size -A /tmp/arduino_build_987778/Blink.ino.elf
O sketch usa 1120 bytes (3%) de espaço de armazenamento para programas. O máximo são 32256 bytes.
Variáveis globais usam 9 bytes (0%) de memória dinâmica, deixando 2039 bytes para variáveis locais. O máximo são 2048 bytes.
../arduino-1.8.16/portable/packages/websim/tools/webuploader/1.0/webuploader.sh /tmp/arduino_build_987778/Blink.ino.hex 
```

urish, 2022-04-19:
Version numbers seem to be the same as in the build server.

You can try with the arduino-cli and see if you get different results:

```
arduino-cli compile -e -b arduino:avr:uno --build-property build.extra_flags=-Og blink
```

ricardojlrufino, 2022-04-19:
Still different
[image]

A kinda basic question, should the ELF address be the same as the (PC) address?

urish, 2022-04-19:
Good question - PC doesn't include the LSB, so you should multiply it by 2. 

And since 0x1af === 0x35e, this is referring to the same address. Sorry for not mentioning this earlier.

ricardojlrufino, 2022-04-19:
Good questions make good answers!! hahaha
Thank you friend

There would already be a theoretical base point to implement a STEP BY STEP, at least looking at the ELF. Transporting this to which file and line there is another challenge!

urish, 2022-04-19:
Try `avr-objdump -WL firmware.elf`, it might be helpful

ricardojlrufino, 2022-04-19:
why did you show me this...

now I want to implement... lol

ricardojlrufino, 2022-04-19:
Although it is not my purpose, to make a debugger, but I wanted to understand...

ricardojlrufino, 2022-04-20:
I couldn't resist trying it out...
https://user-images.githubusercontent.com/515675/164158700-d6b8a35a-7feb-4349-9e69-f69a922259cc.mp4

urish, 2022-04-20:
Cool! Was it hard?

ricardojlrufino, 2022-04-20:
There are still some things that need to be reviewed.
But it was relatively easy so far..

For example, I'm only focused on the main (INO) file.

 for example, two lines appeared in the same address
[image]

As it's a simple blink, I don't know how it would be the case of a For... or While... I didn't get to test it.

The central idea is:
In my .ino file extract all lines (from avr-objdump -WL) and set as breakpoints.

I also get the code of each line and send it, to be able to validate if it was correct. But it would have to be done differently, like sending the entire file.

I'll leave part of the code that makes the parser, maybe it can help

```java
/**
   * Parse result of: avr-objdump -WL 
   */
  private static List<ObjSourceMap> parserObjDumpWL(String fileDumpTxt) throws IOException {

    List<ObjSourceMap> list = new LinkedList<DebugClientHandler.ObjSourceMap>();
    
    String[] lines = fileDumpTxt.split("\n");
    
    final Pattern fileRegEx = Pattern.compile("(.*\\.\\S{1,3}):"); // Math file header
    final Pattern addrRegEx = Pattern.compile("(\\w*\\.\\S{1,3})\\s*(\\d+)\\s*(0.*)"); // math file lines and addr
    
    String currentFile = null;

    for (String line : lines) {
      final Matcher matcherFile = fileRegEx.matcher(line);
      final Matcher matcherAddr = addrRegEx.matcher(line);

      if (matcherFile.find()) {
        currentFile = matcherFile.group(1);
        
        // fix source of ino.. add /sketch/
//        if(currentFile.endsWith("ino")) {
//          Path path = Paths.get(currentFile);
//          Path parent = path.getParent();
//          Path newPath = Paths.get(parent.toString(), "sketch", path.getFileName().toString());
//          currentFile = newPath.toString().replace("modified_sketch", "build");
//        }
        
      }

      while (matcherAddr.find()) {
        ObjSourceMap fileDump = new ObjSourceMap(currentFile, matcherAddr.group(2), matcherAddr.group(3));
        list.add(fileDump);
        System.out.println("fileDump" + fileDump);
      }

    }
    
    grabLineSources(list); // TODO: this can be removed in future
    
    return list;
  }

static class ObjSourceMap{
    String file;
    String lineNumber;
    String addr;
    String lineSource; // source code of line.
 /// ....
 } 
```

On avr8js side:
[image]

lemio, 2022-04-20:
> I couldn't resist trying it out... https://user-images.githubusercontent.com/515675/164158700-d6b8a35a-7feb-4349-9e69-f69a922259cc.mp4

Great work! Love the fact that the milliseconds since start is on the chip as-well! Just out of curiosity, why is it skipping over some lines and highlighting some lines that don't have a clear action? Lines like 

`}` or `setup(){`are highlighted while lines like `digitalWrite(LED_BUILTIN, HIGH);` is skipped.

ricardojlrufino, 2022-04-20:
@lemio 
There are still some bugs that have to be dealt with. I don't know why, but the same address appears on two different lines in the ELF file.
[image]

I just wanted to do a proof of concept to know how difficult it would be to map the simulation in Av8js to the source code... and it proved to be feasible.

urish, 2022-07-13:
This feature request has got to the [top of the vote list](https://wokwi.com/features), so I'm starting to work on it.

Progress so far: a basic ELF parser - https://github.com/wokwi/elfist

urish, 2022-08-26:
Some people asked about the progress, so here's a quick update - 

1. The user interface for setting breakpoints is ready: https://medium.com/@urish/angular-schematics-and-the-future-of-typewiz-c266a5442c82
2. Currently working on deciphering the [DWARF 5 specification](https://dwarfstd.org/doc/DWARF5.pdf). This one is quite challenging, and is one of the reasons why the debugger implementation takes so long.

urish, 2022-09-15:
Some more progress updates: Copilot is helping to [speed things up](https://twitter.com/UriShaked/status/1570156818495868928), but there's still a lot of work to be done.

urish, 2022-09-19:
Early prototype you can start playing with (and giving feedback):

https://debugger.preview.wokwi.com/projects/343260414295933522

Video showing it in action:

https://twitter.com/WokwiMakes/status/1571956642564149248

Note that the preview version currently always compiles the project in Debug mode, so it might take longer to compile, and projects using the FastLED library may break (as FastLED relies on specific compiler optimizations that aren't enabled in debug mode)

## What works?

* Breakpoints
* Step into
* Should work with any Arduino Uno/Nano/Mega (AVR) project, at least in theory

## What doesn't work yet?

* Step over / Step out - working on it right now
* ESP32 / Pi Pico support - coming soon

## What's missing?

You tell me :-)

ali80, 2022-09-19:
how can I watch local and global variables?

urish, 2022-09-20:
> how can I watch local and global variables?

Right now you can't, but I'm thinking about adding a panel where you can see the values of local variables. Globals are a more tricky, as many of them are defined by the Arduino Core libraries, so the list will be pretty long. Any thoughts about a simple solution?

ali80, 2022-09-20:
Maybe don't show the whole global variable list at startup, only show them if the user adds them by name

lemio, 2022-09-20:
Amazing work! Great to see, I think a lot of teachers could use this to explain loops, if else, variables, iteration, etc. in an interactive way; with actual Arduino/Cpp code!

Here are some comments that would make it potentially even more awesome. Don’t see it as critique, just some ideas (none of which are tested yet)
-	At first, I had some problem understanding the concept of debugging. I didn’t put breakpoints and expected the “step to next” icon to just go to the next line. I also would expect that the pressing the “debug pause” icon would show me where we are currently in the code. Afterwards I expect to use the jump to next to jump to the next line.
-	I would visually make the play pause/stop on the right side connected to the stepper on the left side, perhaps just move all debugging elements to the place and style of the other elements.
-	In this example https://debugger.preview.wokwi.com/projects/343326855340753491 it would be great if I could see what lines are being executed; now it only does that when making breakpoints.
-	https://makecode.microbit.org/ ‘s “slow mode” could also be a great feature to show what happens sequentially
https://user-images.githubusercontent.com/877056/191330826-873f871b-c384-4ab6-92a8-48b9d3b573fc.mp4
-	Both of Bret Victors essays on improving programming could be interesting to look at
-	http://worrydream.com/#!/LearnableProgramming
-	http://worrydream.com/#!2/LadderOfAbstraction 
-	Specifically, [this one](http://worrydream.com/LearnableProgramming/Movies/Flow10.mp4) would be nice for the example shared above. Seeing what lines of code are being executed; since there is 3 interactive elements (the breakpoints, play/next pause, and the hardware to interact with it could be nice to first get a glimpse over what code is executed in general.  
-	[This one](http://worrydream.com/LearnableProgramming/Movies/Flow5.mp4) could be interesting since it shows the variable values when hovering over the function.
-	The Stop icon is slightly confusing, I would make it a cross since it is closing the window element (now I would expect the element to stay), in general I would advice to not have two time-controlling areas, but add the debug elements to the one on the right.
-	Tooltips when hovering over the elements, Since I'm not that familiar with debugging I don’t know what dot + arrow up means; after trying it out, it seems to be less relevant for this editor to move up and down layers using these buttons. (often the layer down is not an user file)
-	Hovering over global/local variables could be a great way to choose what to monitor.
-	It still seems to debug non-user code (f.e. it will get stuck at servo.write for 5 steps) and 3 times at the end of loop.
-	Make it clear that code ‘should’ not (or can not) be edited during debug mode
-	Either changing the code exits/restarts the debug mode
-	Or  there could be some visual way to indicate that because changing it it would no longer match. 
https://user-images.githubusercontent.com/877056/191330784-facf9202-fe61-4c83-832e-7aab418a4e4e.mp4

urish, 2022-09-20:
Thanks for the very detailed and thoughtful comment!

In terms of the UX - there's still much to be done. You should consider the debug bar is as a rough draft.

One major point that we need to figure out before going into all the tiny details - how does Wokwi know whether the user wants to run the program in Release or Debug mode. 

The program has to be compiled differently, and we need to know it whenever the simulation starts. For the prototype (at `debugger.preview.wokwi.com`), we currently always assume Debug mode for simplicity. But programs take longer time to compile in Debug mode, they need to download more data (so slower startup time), and some larger programs will get too big and fail to compile in Debug mode. So it can't stay the default.

Once we have a way to start the program in Debug mode, the idea is to show the debug bar (with all the button) only if the program is running in Debug mode. Otherwise, the stepping buttons, etc. make no sense.

Here are two approaches I'm considering:

1. Have a new button for starting a debug session, so the existing green Play button still runs the Release version as it does today in Wokwi.com

2. Automatically run the program in Debug mode if the user has set any breakpoints before hitting the play button. Less UI clutter, but a bit more implicit / "magical". On the other hand, the Debug mode not very useful if you don't set any breakpoints.

Would love to hear more thoughts!

Any thoughts?

urish, 2022-09-21:
Update: We have tooltips for the debugger buttons!

lemio, 2022-09-21:
> * Have a new button for starting a debug session, so the existing green Play button still runs the Release version as it does today in Wokwi.com

This seems more sensible, perhaps triggering the debug mode could start into 'slow' mode, so non-fw-engineers could use this as a way to understand what debugging is, before having to set breakpoints. Both could of course also work, or in some way the debug button could be highlighted in the UI when setting a breakpoint, so the user still has full control and learns the connection between the dots on the left (breakpoints) and the debug option (bug button on the right f.e.).

urish, 2022-09-23:
Update: added "Debug" option in the dropdown menu. The green play button builds and run the code in Release mode (no functional breakpoints, etc.), while the "Debug" option builds the code in debug mode and activates the debugger.

[image]

urish, 2022-09-24:
And now you can also step into system/library functions (`delay()`, `digitalWrite()`, etc.):

[image]

urish, 2022-10-19:
There's an initial interface to see local / global variables:

[image]

The user interface definitely needs some work, but you can already start playing around with it and hunt bugs (there are many, I'm sure)

tyeth, 2022-10-28:
This is unreal, can't wait for esp32 support, and does it also extend to circuit/micropython too? I should upskill, my gdb is too weak :) 

Do we still use the project link at the top or can we do any project just by adding the subdomain prefix? 

Also feedback-wise, as a fairly seasoned software developer (not hardware) I expected to jump out of the delay/timer_ms functions back to the sketch and then be able to continue one line from the end of the for-loop back to the first instruction in the loop, but always continues/goes deeper and only setting breakpoints seems to give me the friendly sketch highlighted line as it is about to be run. An example is this project kindly linked earlier, https://debugger.preview.wokwi.com/projects/343326855340753491 and I set a breakpoint on line 32, and change the potentiometer to max, then resume program, fiddle with knob and when <0 it will pause inside the sketch perfectly and I can see  registers etc under globals (presumably as recognised name and change seems to make sense but my binary is weak) as well as locals, so cool. It did get breakpoint madness syndrome after I stopped and started and tried trapping the pin13 low event, seeming to prefer a breakpoint in print. Restarting also gets me a breakpoint in pinMode for every pinchange presumably, but I set one and unset it to solve that however my print one is still there instead of pausing in the sketch on line 38. 
[image]

Also keep up the amazing work!

tyeth, 2022-11-24:
It's working well, and loving the alpha local vscode version ;) Just while browsing someone elses project I added the debugger.preview subdomain and found I can't resize the debug panel, and it's painfully small on my rotated monitor :)
[image]

urish, 2022-11-25:
Good to see it in action :) Yeah, we definitely need to work on the user interface for the built-in debugger. Right now, we've shifted the focus to #414, as it will also provide a good debugging solution. But once that is out, we'll keep improving on the web debugger as well (and hopefully release it too).

urish, 2023-02-17:
Updates:

1. The debugger has now been merged into wokwi.com, as a beta feature
2. The debugger pane is now fully resizable (cc @tyeth). 
3. Only AVR targets (Arduino Uno, Nano, Mega and ATtiny85) are supported. 

The debugger is not started by default. To start a debug session, choose the "Debug (beta)" option from the drop down menu in the diagram editor:

[image]

urish, 2023-03-20:
Initial docs are now available too: https://docs.wokwi.com/guides/debugger

closed 2023-03-20 by urish

traisjames, 2023-09-19:
I am finding my user defined variables are just showing as <unavailable>.  The code still works, in this case I expected the variable 'mode' to be 11 and it stepped into the case statement correct.
[image]

urish, 2023-09-19:
> I am finding my user defined variables are just showing as . Based on the Case statement, the variable 'mode' has 11, it should be 12.

Can you please share a link to the project in question?

In most cases, the compiler does optimizations that some of the variables unavailable in runtime. Does `mode` show as unavailable even on the line that says `switch (mode) {` ?

traisjames, 2023-09-19:
> In most cases, the compiler does optimizations that some of the variables unavailable in runtime. Does `mode` show as unavailable even on the line that says `switch (mode) {` ?

Woops I was loosing track of my 11 and 12, it is behaving as it should, I just had a bug in my code.
Yes I just check and it still shows `byte mode = <unavailable>`

The project is at https://wokwi.com/projects/376272565810605057

urish, 2023-09-19:
> Yes I just check and it still shows `byte mode = <unavailable>`

Thanks, investigating

urish, 2023-09-19:
@traisjames uploaded a fix, can you please check again and report?

traisjames, 2023-09-19:
> @traisjames uploaded a fix, can you please check again and report?

Yes it now reads as I would expect it to.  Thank you.

gsuberland, 2024-02-14:
There seems to be a bug when displaying values of local variables, or resolving their addresses. When debugging [this ATTiny85 sketch](https://wokwi.com/projects/389749986496967681), the debugger shows float locals as `13073.3740234375`. If you convert this IEEE754 number back into hex, you get `0x464c457f`, which is the signature of an ELF executable. The integer locals show consistently incorrect values, so presumably they're also being resolved to the wrong address.