wokwi/wokwi-features #208: Improved mobile experience

StevenMartin1, 2021-10-16:
Is it possible for the mobile version of the site to have an improved experience?

I dont intend on exclusively using the site from mobile, but to jot down code ideas when they happen at a moments notice or to quickly hop on the change/update something would be great. I think minimizing buttons for the file tabs and the arduino window would be good to maximise the code space on a small screen. 

Not sure if its just me but im currently using a Samsung S20 with the Swiftkey keyboard, its usable but quite problematic. Such as i cant copy paste etc

Anyway great work! I have a few more ideas i may also post.

urish, 2021-10-16:
Thanks for sharing your feedback here Steven!

We currently have an initial prototype for new mobile layout: https://new-mobile.preview.wokwi.com/arduino/new

There're still some rough edges, especially when you zoom in/out in the diagram. @TimMathias is trying to figure out that part.

As for editing code on mobile - the code editor that we use, [monaco-editor](https://microsoft.github.io/monaco-editor/), does not have very good mobile support. Quoting from their home page:

> The Monaco editor is not supported in mobile browsers or mobile web frameworks.

One possible solution would be to use a different editor, such as [CodeMirror](https://codemirror.net/), but that would require some adaptation (e.g. rewriting the syntax highlighting engine, a different approach to display compilation errors, etc).

I opened this feature request [for voting](https://wokwi.com/features#feature-1028025160). Once it goes up the list I will start looking at integrating CodeMirror or a similar solution for editing code on mobile.

djedu28, 2021-11-20:
Hello.

I come with my little contribution to the subject.

It's true that the editor doesn't support touch screens, and the keyboard on mobile devices is the main bottleneck.

But the current editor supports keyboard shortcuts even on mobile splints.

I learned to program using a cell phone (since I didn't have a computer), so the solutions I used to get around the barriers were:

Using other keyboarding applications (usually more than one, and switching between them as something specific is needed)
My recommendations are:

- Gboard
https://play.google.com/store/apps/details?id=com.google.android.inputmethod.latin&hl=pt_BR&gl=US
[image]

- CodeBoard Keyboard for Coding
https://play.google.com/store/apps/details?id=com.gazlaws.codeboard
[image]

- Hacker's Keyboard
https://play.google.com/store/apps/details?id=org.pocketworkstation.pckeyboard
[image]

Using Gboard as main (with autocomplete disabled)
and when I program needing the tab and arrows, I use CodeBoard right away.

And when I need to use keyboard shortcuts (CTRL, alt, shift, ...) a lot, then I use Hacker's Keyboard.

As said before, even on the mobile screen, I can still use shortcuts like tab to activate auto complete, CTRL+Z to undo and using shift+arrows I can select, cut-and-paste everything on the keyboard.

Ps.: To select, cut, paste I prefer the Gboard, in the selection menu:
[image]

djedu28, 2021-11-20:
My other tool to get around the barriers to programming on the smartphone was to use a simple text editor, which supported text markup. Programming using it, and at the end copying and pasting the code where necessary (in this case, program with it and then copy it to wokwi). Because, using these offline editors, the touch screen shortcuts are functional, and together with the virtual keyboard hint, the programming productivity sometimes even surpasses that achieved using a computer with a physical keyboard and mouse.

For this I recommend the editors:
- DroidEdit Free
https://play.google.com/store/apps/details?id=com.aor.droidedit
[image]
[image]

or the
- QuickEdit  (I used it a lot in the past and recommended it to friends)
https://play.google.com/store/apps/details?id=com.rhmsoft.edit
[image]

and if you have a good smartphone and want the smart autocomplete feature, i recommend the IDE called ACODE. (Program even REACTjs using it).

- Acode - powerful code editor
https://play.google.com/store/apps/details?id=com.foxdebug.acodefree
[image]

urish, 2021-12-12:
Thanks @djedu28 for sharing this! 

Looking at ACode, they seem to use the [ACE editor](https://ace.c9.io/) behind the scenes.

Which of these editors do you use most often?

djedu28, 2022-01-14:
@urish what I use the most to edit is DroidEdit Free, for being light, including copying and pasting text in/to the wokwi.

I rarely use ACode, only when I'm away from the computer and needing to program and test projects that need to run a local server.

disappear123df, 2023-01-15:
thanks looking this help me able to select using the text editor

djedu28, 2023-05-06:
I'm returning to this post to bring some information I forgot.

When accessing an old smartphone of mine, I found a much better IDE than the ones I mentioned, which when I changed smartphones ended up getting lost because I couldn't find it in the new smartphone's app store.
And over the years I ended up thinking it was a mental delusion, remembering such a complete IDE on an Android.

I used this ide in 2019, unfortunately it has been discontinued, however it still works great.

The name is I<code>, and when I mentioned "Acode" I was quoting my memories of using "i<code>"

https://apk.support/app-pt/com.twoseventwodev.icodego