wokwi/wokwi-features #213: I2S and DMA support for ESP32

costyn, 2021-10-20:
**Is your feature request related to a problem? Please describe.**
It would be cool if we can use the FastLED clockless I2S/DMA driver on ESP32, so we can test it against "real world" code.

Also, since it seems the FastLED on ESP32 with RMT is broken, this might be a nice alternative.

**Describe the solution you'd like**
I would like if my I2S test project works.
https://wokwi.com/arduino/projects/312993650809242176

urish, 2022-09-11:
This issue has been getting a lot of votes recently, and I'd love to learn why people are voting for it.

Do you need I2S on the ESP32 for:

1. Audio output
2. Audio input
3. FastLED
4. Other purpose (please specify)

Please add your comments to this discussion - they'll help prioritizing our work.

costyn, 2022-09-11:
I have been using my vote-powers on this issue since I first got them :) Not sure who else has been voting, but I need it for FastLED.

urish, 2022-09-11:
> but I need it for FastLED

Can you share a link to a Wokwi project that has the setup you'd like to use? (FastLED + I2S)?

costyn, 2022-09-11:
https://wokwi.com/projects/315542185092055618  For example. The real-world project using part of these algorithms are already using I2S. But well, I'm not working much on the project these days. I don't foresee the need to simulate it in Wokwi in the near future. So for me I2S doesn't really have that much priority at the moment. I don't know how much work this would be to implement.

jackdewhurst, 2023-12-28:
Would love support for FastLED please 🙏

urish, 2023-12-28:
The good news is that the DMA part is already complete. Leaving just the I2S part to be implemented :)

SleepInfinity, 2024-12-24:
audio input and output would be so cool !

baranovmv, 2025-01-29:
Both Audio input and output, via DMA