wokwi/wokwi-features #222: Annotation

kvapehe, 2021-10-30:
Potentially add a en element named annotation.
A code or project cannot be to much commented or annotated.

Here is an example. I added the "circuit" DHT22
And yes DHT is an abbreviation of the text below.
Digital Humidity and Temperature sensor.
I also wanted to add schematic information in the drawing.
So annotation would be a nice feature.
And colouring wires the same.
If annotation is added, then do some thinking how to add it and what features it should have.
Title, different colours etc, and how it should be a part of the project. 
(If it is attached to the canvas or a component etc.)

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
No problem, just a missed feature, and not important.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
See above for ideas.

Sorry if this is not nicely formatted. Maybe the author will rewrite it before publishing 
and making it available for voting. This was not an feature request, more like planting an idea.
If a similar suggestion is posted earlier, sorry for the "request" I searched for words synonymous to annotation 
and comments.

Regards

urish, 2021-10-30:
Thanks for the suggestion @kvapehe!

Is it different from #43 ?

djedu28, 2021-10-30:
for now you can use the "README.md" file inside the wokwi project.

Using the same pattern used in Github 

https://docs.github.com/en/repositories/managing-your-repositorys-settings-and-features/customizing-your-repository/about-readmes

You can add images, illustrations, tables, ... and make understanding the project more didactic.

Check the wokwi sample projects, they already have the README.md file

kvapehe, 2021-10-30:
Hi, where do I answer to his.
I hope reply with email does the trick.

No, not much different, but, there is one difference.

The annotation is not something added to a component, but may be added.
Also on the Canvas. I some sketches ( tinkercad or pspice or, others.

You may want to add a general comment. Mostly these a re coloured (grey / yellow / white)
But somethimes you want the attention and you may use more colours.
So I was thinking when wwires are colured in the drawing-tool this functioin could also apply to comments.

I did not se nr 43 as I did not search for text or label / lable.
If I had seen it I would have refered t it and added suggestions.
And as written earlier, it is not a request. I just missed it when I found the info on the doc, got back to the drawing and did not remeber the information. If I had an annotation I would have copy/pased the interesting information.

The nr 43 does this a long way, but is more like when you hover a LED it tells you it is LED, a maybe the pins too.

The one I was think of was add the info I need as the designer or as the user afterwars.

Like this circuit with two Arduinos communicate with Tx/Rx and a baudrate, then telling th user \n is used as line feed code.
Any arbitrary comment / annotation / text you may think of.

I used Makecode from MS with Micobit, and thay have comment / annotation as an option.
If you simply add it th the programming it attaches to the canvas (block-coding) If you add it to an element it becomes inline comment. (I know this is the code, but for the user those blocs are like the circuits we see on wokwi.

I hope this eleborated a little, and if 43 is woted it may appear sometime.

Thanks for reaching out way faster than what I expected.

Regards

https://docs.github.com/en/repositories/managing-your-repositorys-settings-and-features/customizing-your-repository/about-readmes

closed 2022-08-26 by urish