wokwi/wokwi-features #223: Translation update: (New texts #221) — Portuguese - Text

djedu28, 2021-10-30:
Below is an update of the wokwi translation (pt_BR)

pt_BR.txt
```
userMenu.signIn:
- source: Sign in
- translation: Entrar

userMenu.signUp:
- source: Sign up
- translation: Inscrever-se

userMenu.discord:
- source: Discord
- translation: Discord

userMenu.myProjects:
- source: My projects
- translation: Meus projetos

userMenu.theClub:
- source: The Club
- translation: O Clube

userMenu.featureRoadmap:
- source: Feature Roadmap
- translation: Roteiro de Recursos

userMenu.logout:
- source: Logout
- translation: Sair

project.header.save:
- source: Save
- translation: Salvar

project.header.share:
- source: Share
- translation: Compartilhar

project.header.likeProjectTooltip:
- source: Like this project
- translation: Curtir este projeto

project.header.unlikeProjectTooltip:
- source: Unlike this project
- translation: Deixar de curtir

project.menu.saveACopy:
- source: Save a copy
- translation: Salvar uma cópia

project.menu.lockProject:
- source: Lock project
- translation: Bloquear projeto

project.menu.unlistProject:
- source: Unlist project
- translation: Remover projeto da lista

project.menu.downloadProjectZip:
- source: Download project ZIP
- translation: Baixar ZIP do projeto

project.error.saveFailed:
- source: Save failed: project is locked!
- translation: Falha ao salvar: o projeto está bloqueado!

project.error.signInToLikeProject:
- source: Please sign-in to Like this project.
- translation: Faça login para curtir este projeto.

editor.fileMenu.rename:
- source: Rename
- translation: Renomear

editor.fileMenu.delete:
- source: Delete
- translation: Excluir

editor.fileMenu.newFile:
- source: New file...
- translation: Novo arquivo...

editor.fileMenu.uploadFile:
- source: Upload file(s)...
- translation: Enviar arquivo(s)...

editor.fileMenu.confirmDelete:
- source: This will remove {{filename}}. Should I go ahead?
- translation: Isso removerá {{filename}}. Deseja prosseguir?

editor.fileMenu.error.invalidFileName:
- source: Invalid filename: character {{char}} is not allowed.
- translation: Nome de arquivo inválido: o caractere {{char}} não é permitido.

editor.fileMenu.error.invalidFileType:
- source: Invalid file type. Supported file types: {{types}}.
- translation: Tipo de arquivo inválido. Tipos de arquivo suportados: {{types}}.

editor.fileMenu.error.fileExists:
- source: A file with this name already exists
- translation: Já existe um arquivo com este nome

editor.fileMenu.error.uploadFailed:
- source: Some files could not be uploaded!
- translation: Alguns arquivos não puderam ser carregados!

diagram.menu.fit:
- source: Fit
- translation: Ajustar

diagram.menu.zoomIn:
- source: Zoom in
- translation: Zoom mais

diagram.menu.zoomOut:
- source: Zoom out
- translation: Zoom menos

diagram.menu.toggleGrid:
- source: Toggle grid
- translation: Alternar grade

diagram.menu.fullScreen:
- source: Full screen
- translation: Tela cheia

diagram.menu.exitFullScreen:
- source: Exit full screen
- translation: Sair da tela cheia

diagram.menu.help:
- source: Help
- translation: Ajuda

diagram.addPart.wokwi-arduino-uno:
- source: Arduino Uno
- translation: Arduino Uno

diagram.addPart.wokwi-arduino-mega:
- source: Arduino Mega 2560
- translation: Arduino Mega 2560

diagram.addPart.wokwi-arduino-nano:
- source: Arduino Nano
- translation: Arduino Nano

diagram.addPart.wokwi-pi-pico:
- source: Raspberry Pi Pico
- translation: Raspberry Pi Pico

diagram.addPart.wokwi-nano-rp2040-connect:
- source: Arduino Nano RP2040 Connect
- translation: Arduino Nano RP2040 Connect

diagram.addPart.wokwi-attiny85:
- source: ATtiny85
- translation: ATtiny85

diagram.addPart.wokwi-led:
- source: LED
- translation: LED

diagram.addPart.wokwi-pushbutton:
- source: Pushbutton
- translation: Botão (Pushbutton)

diagram.addPart.wokwi-resistor:
- source: Resistor
- translation: Resistor

diagram.addPart.wokwi-slide-switch:
- source: Slide switch
- translation: Interruptor Deslizante

diagram.addPart.wokwi-membrane-keypad:
- source: Keypad
- translation: Teclado

diagram.addPart.wokwi-logic-analyzer:
- source: Logic Analyzer (8 channels)
- translation: Analisador Lógico (8 canais)

diagram.addPart.wokwi-max7219-matrix:
- source: LED Dot Matrix (MAX7219)
- translation: Matriz de LEDs (MAX7219)

diagram.addPart.wokwi-led-ring:
- source: NeoPixel Ring
- translation: NeoPixel Ring (Anel de LEDs)

diagram.addPart.wokwi-neopixel-meter:
- source: NeoPixel Meter
- translation: NeoPixel Meter

diagram.addPart.wokwi-buzzer:
- source: Buzzer
- translation: Buzzer

diagram.addPart.wokwi-dht22:
- source: DHT22
- translation: DHT22 – Sensor de Temperatura e Umidade

diagram.addPart.wokwi-potentiometer:
- source: Potentiometer
- translation: Potenciômetro

diagram.addPart.wokwi-slide-potentiometer:
- source: Slide Potentiometer
- translation: Potenciômetro Deslizante

diagram.addPart.wokwi-lcd1602:
- source: LCD 16x2
- translation: LCD 16x2

diagram.addPart.wokwi-lcd1602-i2c:
- source: LCD 16x2 (I2C)
- translation: LCD 16x2 (I2C)

diagram.addPart.wokwi-lcd2004:
- source: LCD 20x4
- translation: LCD 20x4

diagram.addPart.wokwi-lcd2004-i2c:
- source: LCD 20x4 (I2C)
- translation: LCD 20x4 (I2C)

diagram.addPart.wokwi-servo:
- source: Servo
- translation: Servo

diagram.addPart.wokwi-rotary-dialer:
- source: Rotary Dialer
- translation: Discador Rotativo

diagram.addPart.wokwi-ssd1306:
- source: SSD1306 OLED display
- translation: Display OLED SSD1306

diagram.addPart.wokwi-ds1307:
- source: DS1307 RTC
- translation: Relógio RTC DS1307

diagram.addPart.wokwi-hc-sr04:
- source: HC-SR04 Ultrasonic Distance Sensor
- translation: Sensor Ultrassônico  HC-SR04

diagram.addPart.wokwi-rgb-led:
- source: RGB LED
- translation: LED RGB

diagram.addPart.wokwi-7segment:
- source: Seven Segment Display
- translation: Display de Sete Segmentos (1 dígito)

diagram.addPart.wokwi-7segment-4digits:
- source: Seven Segment Display (4 digits)
- translation: Display de Sete Segmentos (4 dígitos)

diagram.addPart.wokwi-led-bar-graph:
- source: LED Bar Graph
- translation: Gráfico de Barras LED

diagram.addPart.wokwi-analog-joystick:
- source: Analog Joystick
- translation: Joystick Analógico

diagram.addPart.wokwi-ir-receiver:
- source: IR Receiver
- translation: Receptor IR

diagram.addPart.wokwi-ir-remote:
- source: IR Remote
- translation: Controle Remoto IR

diagram.addPart.wokwi-pir-motion-sensor:
- source: PIR Motion Sensor
- translation: Sensor de Movimento PIR

diagram.addPart.wokwi-ntc-temperature-sensor:
- source: Analog Temperature Sensor (NTC)
- translation: Sensor Analógico de Temperatura (NTC)

diagram.addPart.wokwi-74hc595:
- source: 74HC595 Shift Register
- translation: 74HC595 CI Registrador de Deslocamento

diagram.addPart.wokwi-franzininho:
- source: Franzininho
- translation: Franzininho

diagram.addPart.wokwi-photoresistor-sensor:
- source: Photoresistor (LDR) Sensor
- translation: Fotorresistor (LDR)

diagram.addPart.wokwi-ky-040:
- source: KY-040 Rotary Encoder
- translation: Encoder Rotativo KY-040

diagram.addPart.wokwi-mpu6050:
- source: MPU6050 Accelerometer + Gyroscope
- translation: MPU6050 Acelerômetro + Giroscópio

diagram.addPart.wokwi-microsd-card:
- source: microSD card
- translation: Cartão microSD

```

urish, 2021-10-30:
Thanks Eduardo! 

@arcostasi any thoughts about the text changes?

Here's the diff from what we currently have:

```diff
--- pt-BR.txt       2021-10-31 01:04:38.092887700 +0200
+++ pt-BR-223.txt       2021-10-31 01:04:44.280579500 +0200
@@ -16,11 +16,11 @@

 userMenu.theClub:
 - source: The Club
-- translation: O clube
+- translation: O Clube

 userMenu.featureRoadmap:
 - source: Feature Roadmap
-- translation: Roadmap de recursos
+- translation: Roteiro de Recursos

 userMenu.logout:
 - source: Logout
@@ -124,7 +124,7 @@

 diagram.menu.exitFullScreen:
 - source: Exit full screen
-- translation:
+- translation: Sair da tela cheia

 diagram.menu.help:
 - source: Help
@@ -160,7 +160,7 @@

 diagram.addPart.wokwi-pushbutton:
 - source: Pushbutton
-- translation: Pushbutton
+- translation: Botão (Pushbutton)

 diagram.addPart.wokwi-resistor:
 - source: Resistor
@@ -172,7 +172,7 @@

 diagram.addPart.wokwi-membrane-keypad:
 - source: Keypad
-- translation: Keypad
+- translation: Teclado

 diagram.addPart.wokwi-logic-analyzer:
 - source: Logic Analyzer (8 channels)
@@ -180,11 +180,11 @@

 diagram.addPart.wokwi-max7219-matrix:
 - source: LED Dot Matrix (MAX7219)
-- translation: LED Dot Matrix (MAX7219)
+- translation: Matriz de LEDs (MAX7219)

 diagram.addPart.wokwi-led-ring:
 - source: NeoPixel Ring
-- translation: NeoPixel Ring
+- translation: NeoPixel Ring (Anel de LEDs)

 diagram.addPart.wokwi-neopixel-meter:
 - source: NeoPixel Meter
@@ -196,7 +196,7 @@

 diagram.addPart.wokwi-dht22:
 - source: DHT22
-- translation: DHT22
+- translation: DHT22 – Sensor de Temperatura e Umidade

 diagram.addPart.wokwi-potentiometer:
 - source: Potentiometer
@@ -204,7 +204,7 @@

 diagram.addPart.wokwi-slide-potentiometer:
 - source: Slide Potentiometer
-- translation: Potenciômetro de Slide
+- translation: Potenciômetro Deslizante

 diagram.addPart.wokwi-lcd1602:
 - source: LCD 16x2
@@ -236,23 +236,23 @@

 diagram.addPart.wokwi-ds1307:
 - source: DS1307 RTC
-- translation: DS1307 RTC
+- translation: Relógio RTC DS1307

 diagram.addPart.wokwi-hc-sr04:
 - source: HC-SR04 Ultrasonic Distance Sensor
-- translation: Sensor Ultrassônico HC-SR04
+- translation: Sensor Ultrassônico  HC-SR04

 diagram.addPart.wokwi-rgb-led:
 - source: RGB LED
-- translation: RGB LED
+- translation: LED RGB

 diagram.addPart.wokwi-7segment:
 - source: Seven Segment Display
-- translation: Display de Sete Segmentos
+- translation: Display de Sete Segmentos (1 dígito)

 diagram.addPart.wokwi-7segment-4digits:
 - source: Seven Segment Display (4 digits)
-- translation: Display de Sete Segmentos (4 digitos)
+- translation: Display de Sete Segmentos (4 dígitos)

 diagram.addPart.wokwi-led-bar-graph:
 - source: LED Bar Graph
@@ -280,7 +280,7 @@

 diagram.addPart.wokwi-74hc595:
 - source: 74HC595 Shift Register
-- translation: 74HC595 Shift Register
+- translation: 74HC595 CI Registrador de Deslocamento

 diagram.addPart.wokwi-franzininho:
 - source: Franzininho
@@ -288,11 +288,11 @@

 diagram.addPart.wokwi-photoresistor-sensor:
 - source: Photoresistor (LDR) Sensor
-- translation: Sensor fotorresistor (LDR)
+- translation: Fotorresistor (LDR)

 diagram.addPart.wokwi-ky-040:
 - source: KY-040 Rotary Encoder
-- translation: Codificador Rotativo KY-040
+- translation: Encoder Rotativo KY-040

 diagram.addPart.wokwi-mpu6050:
 - source: MPU6050 Accelerometer + Gyroscope
```

djedu28, 2021-10-31:
I appreciate the opportunity to contribute to the project.

@arcostasi Thanks for the initial translation, I await your review. [Obrigado pela tradução inicial, espero tua revisão]

## I can summarize my updates in 5 cases

### *1 — In some updates, I've set the capitalization style to headline mode;

```diff
- - translation: O clube
+ - translation: O Clube
``` 

### *2 — Some translations were still in English, and I translated them;

```diff
 userMenu.featureRoadmap:
- - translation: Roadmap de recursos
+ - translation: Roteiro de Recursos

 diagram.menu.exitFullScreen:
 - source: Exit full screen
- - translation: 
+ - translation: Sair da tela cheia

 diagram.addPart.wokwi-max7219-matrix:
 - source: LED Dot Matrix (MAX7219)
- - translation: LED Dot Matrix (MAX7219)
+ - translation: Matriz de LEDs (MAX7219)

 diagram.addPart.wokwi-slide-potentiometer:
 - source: Slide Potentiometer
- - translation: Potenciômetro de Slide
+ - translation: Potenciômetro Deslizante

``` 

### *3 — Some components were using a name not accessible to those new to electronics/robotics, so I upgraded to the more popular name;

```diff

 diagram.addPart.wokwi-pushbutton:
 - source: Pushbutton
- - translation: Pushbutton
+ - translation: Botão (Pushbutton)

 diagram.addPart.wokwi-membrane-keypad:
 - source: Keypad
- - translation: Keypad
+ - translation: Teclado

 diagram.addPart.wokwi-led-ring:
 - source: NeoPixel Ring
- - translation: NeoPixel Ring
+ - translation: NeoPixel Ring (Anel de LEDs)

 diagram.addPart.wokwi-dht22:
 - source: DHT22
- - translation: DHT22
+ - translation: DHT22 – Sensor de Temperatura e Umidade
 
 diagram.addPart.wokwi-ds1307:
 - source: DS1307 RTC
- - translation: DS1307 RTC
+ - translation: Relógio RTC DS1307
 
 diagram.addPart.wokwi-rgb-led:
 - source: RGB LED
- - translation: RGB LED
+ - translation: LED RGB

 diagram.addPart.wokwi-7segment:
 - source: Seven Segment Display
- - translation: Display de Sete Segmentos
+ - translation: Display de Sete Segmentos (1 dígito)
 
 diagram.addPart.wokwi-74hc595:
 - source: 74HC595 Shift Register
- - translation: 74HC595 Shift Register
+ - translation: 74HC595 CI Registrador de Deslocamento

 diagram.addPart.wokwi-photoresistor-sensor:
 - source: Photoresistor (LDR) Sensor
- - translation: Sensor fotorresistor (LDR)
+ - translation: Fotorresistor (LDR)
 
 diagram.addPart.wokwi-ky-040:
 - source: KY-040 Rotary Encoder
- - translation: Codificador Rotativo KY-040
+ - translation: Encoder Rotativo KY-040
``` 

### *4 — typo

Accent in "dígitos" 
```diff 
 diagram.addPart.wokwi-7segment-4digits:
 - source: Seven Segment Display (4 digits)
- - translation: Display de Sete Segmentos (4 digitos)
+ - translation: Display de Sete Segmentos (4 dígitos)
```

### *5 —  I accidentally put an extra space.

```diff 
 diagram.addPart.wokwi-hc-sr04:
 - source: HC-SR04 Ultrasonic Distance Sensor
- - translation: Sensor Ultrassônico HC-SR04
+ - translation: Sensor Ultrassônico  HC-SR04
```

arcostasi, 2021-11-01:
> @urish I appreciate the opportunity to contribute to the project.
> 
> @arcostasi Thanks for the initial translation, I await your review. [Obrigado pela tradução inicial, espero tua revisão]
> 
> ## I can summarize my updates in 5 cases
> ### *1 — In some updates, I've set the capitalization style to headline mode;
> ```diff
> - - translation: O clube
> + - translation: O Clube
> ```
> 
> ### *2 — Some translations were still in English, and I translated them;
> ```diff
>  userMenu.featureRoadmap:
> - - translation: Roadmap de recursos
> + - translation: Roteiro de Recursos
> 
>  diagram.menu.exitFullScreen:
>  - source: Exit full screen
> - - translation: 
> + - translation: Sair da tela cheia
> 
>  diagram.addPart.wokwi-max7219-matrix:
>  - source: LED Dot Matrix (MAX7219)
> - - translation: LED Dot Matrix (MAX7219)
> + - translation: Matriz de LEDs (MAX7219)
> 
>  diagram.addPart.wokwi-slide-potentiometer:
>  - source: Slide Potentiometer
> - - translation: Potenciômetro de Slide
> + - translation: Potenciômetro Deslizante
> ```
> 
> ### *3 — Some components were using a name not accessible to those new to electronics/robotics, so I upgraded to the more popular name;
> ```diff
>  diagram.addPart.wokwi-pushbutton:
>  - source: Pushbutton
> - - translation: Pushbutton
> + - translation: Botão (Pushbutton)
> 
>  diagram.addPart.wokwi-membrane-keypad:
>  - source: Keypad
> - - translation: Keypad
> + - translation: Teclado
> 
>  diagram.addPart.wokwi-led-ring:
>  - source: NeoPixel Ring
> - - translation: NeoPixel Ring
> + - translation: NeoPixel Ring (Anel de LEDs)
> 
>  diagram.addPart.wokwi-dht22:
>  - source: DHT22
> - - translation: DHT22
> + - translation: DHT22 – Sensor de Temperatura e Umidade
>  
>  diagram.addPart.wokwi-ds1307:
>  - source: DS1307 RTC
> - - translation: DS1307 RTC
> + - translation: Relógio RTC DS1307
>  
>  diagram.addPart.wokwi-rgb-led:
>  - source: RGB LED
> - - translation: RGB LED
> + - translation: LED RGB
> 
>  diagram.addPart.wokwi-7segment:
>  - source: Seven Segment Display
> - - translation: Display de Sete Segmentos
> + - translation: Display de Sete Segmentos (1 dígito)
>  
>  diagram.addPart.wokwi-74hc595:
>  - source: 74HC595 Shift Register
> - - translation: 74HC595 Shift Register
> + - translation: 74HC595 CI Registrador de Deslocamento
> 
>  diagram.addPart.wokwi-photoresistor-sensor:
>  - source: Photoresistor (LDR) Sensor
> - - translation: Sensor fotorresistor (LDR)
> + - translation: Fotorresistor (LDR)
>  
>  diagram.addPart.wokwi-ky-040:
>  - source: KY-040 Rotary Encoder
> - - translation: Codificador Rotativo KY-040
> + - translation: Encoder Rotativo KY-040
> ```
> 
> ### *4 — typo
> Accent in "dígitos"
> 
> ```diff
>  diagram.addPart.wokwi-7segment-4digits:
>  - source: Seven Segment Display (4 digits)
> - - translation: Display de Sete Segmentos (4 digitos)
> + - translation: Display de Sete Segmentos (4 dígitos)
> ```
> 
> ### *5 — I accidentally put an extra space.
> ```diff
>  diagram.addPart.wokwi-hc-sr04:
>  - source: HC-SR04 Ultrasonic Distance Sensor
> - - translation: Sensor Ultrassônico HC-SR04
> + - translation: Sensor Ultrassônico  HC-SR04
> ```

Obrigado @djedu28, ficou ótimo da forma como fez! All right, you can update ;)

urish, 2021-11-02:
Excellent, texts updated! 

Muito obrigado :-)

closed 2021-11-02 by urish