wokwi/wokwi-features #224: Unable to add probonopd/TimeLord library via libraries.txt nor by file upload

DavidThijs, 2021-11-06:
**Is your feature request related to a problem? Please describe.**

Attempted to add TimeLord to the libraries.txt file but it results in a compile error
Attempted to add Timelord.h to the files but the library can still not be found
Source: https://github.com/probonopd/TimeLord

**Describe the solution you'd like**
A possibility to reference less known or popular libraries by adding them to the set of files

leftCoast, 2021-11-07:
It doesn't look like this was ever added to the Arduino IDE library list. Does the libraries.txt list feature work on libraries NOT on the IDE list?

urish, 2021-11-07:
libraries.txt only works for libraries that are available in the Arduino Library Manager.

You can still use the library by copying the files into your project. In this case, you'll have to change the #include statements to use " " instead of < >. Here's an example for the [TimeLord](https://github.com/probonopd/TimeLord) library:

https://wokwi.com/arduino/projects/314584182162457154

DavidThijs, 2021-11-07:
Thanks a lot, including the libraries with the "" instead of the <> did the job !

closed 2021-11-07 by DavidThijs