wokwi/wokwi-features #225: Support ESP32 Bluetooth low energy (BLE)

uriyacovy, 2021-11-08:
**Is your feature request related to a problem? Please describe.**
Many esp32 projects rely on BLE to communicate and configure the device. Wokwi support for BLE would be really useful.

**Describe the solution you'd like**
Implementation of either circuitpython's BLERadio or micropython's BLE modules.

DigiDwrf, 2023-10-24:
It would be great to test code that emulates bluetooth HID devices such as keyboards, mouses, and gamepads.

msn444, 2024-01-15:
Ideal solution would be to pass the host PC BLE interface through to the simulated ESP32, so that the simulated ESP32 can create a real GATT server that can be accessed by nearby devices.

Right now, it seems that attempts to initialize the BLE controller (calls to esp_nimble_hci_and_controller_init()) cause the simulation to hang. In the interim it would be nice if the BLE initialization failed gracefully; right now we need extra code to figure out if it's running in Wokwi so that it doesn't try to init BLE and hang.

urish, 2024-02-04:
> Right now, it seems that attempts to initialize the BLE controller (calls to esp_nimble_hci_and_controller_init()) cause the simulation to hang. In the interim it would be nice if the BLE [...]

Thanks for the feedback! Do you have a minimal test case that reproduces this? Perhaps one of the esp-idf examples?

jumpjack, 2024-03-14:
any news on the subject?

urish, 2024-03-14:
> any news on the subject?

No news. Simulating BLE is a lot of work, and the main use case we're seeing is provisioning for WiFi credentials. If we see interesting commercial use cases, then it may help prioritizing this.

samufnasc, 2025-04-03:
It's very important to simulate bluetooth at this application. I am a teacher at school and I needed to do a project with my class, and I could not use bluetooth to control servo motors. So, I hope this option will be available soon.

JT1960, 2026-11-10:
+1, with a concrete use case.

I’m building a DIY force handle : a XIAO ESP32-S3 reads a load cell through a NAU7802 (I²C, 320 samples/s) and streams the data over BLE to an iPhone app. The firmware is NimBLE-Arduino: one custom 128-bit GATT service with a notify characteristic (38-byte packets, 40 notifications/s), a write characteristic for commands (TARE, START, STOP, SLEEP) and a read characteristic (calibration, firmware version, battery).

Wokwi already covers the rest of my bench very well. My I²C and serial firmware runs unchanged on wokwi.com, and in CI through wokwi-cli and GitHub Actions, with a custom NAU7802 chip driven by set-control. BLE is the only part I still have to test on real hardware.

Even a minimal BLE peripheral simulation would help a lot:

advertising with a custom device name (e.g. C2AW-ROW-01);
a GATT server: services, read/write/notify characteristics, MTU negotiation;
a way to act as the central from outside the simulation. For example, automation scenario steps such as ble-connect, ble-write and ble-expect-notify (an expected value or a payload length), or a bridge to a host-side client such as Python bleak through the Private IoT Gateway.

Classic Bluetooth, coexistence with Wi-Fi and radio-level timing are not needed for this use case. Being able to check in CI that the GATT service is exposed correctly and that notifications arrive at the expected rate would already catch most regressions.

Thanks for Wokwi, it has become part of my test bench.

urish, 2026-11-11:
Thanks for the concrete use case, it's really useful @JT1960.

We have actually started working on BLE simulation, and have a working BLE prototype (connect, advertise, GATT read/write/notify) on ESP32-C3/C6. On the S3, advertising works, but connecting is still in the works. NimBLE-Arduino builds for the C3 and C6 too. Would you be up for testing an early version on a C3 or C6 board while we finish the S3?

JT1960, 2026-11-11:
Thanks a lot @urish, great news!
Yes, I'd be glad to test it. 
Our BLE firmware stage is a few weeks away (we're finishing the acquisition side first), so I could test on a C3 or C6 as soon as that stage starts, unless the S3 is ready by then.
Our use case is a small NimBLE-Arduino peripheral:- one custom GATT service with 3 characteristics: Force (notify), Command (write: TARE, RATE, START, STOP) and Info (read)- notifications of 38-byte packets (8 samples + sequence number + timestamp) at 40 packets/s
Two questions to plan ahead:
1. What plays the central role in the simulation: a built-in client, Web Bluetooth from the browser, or something else?
2. Will BLE be usable from wokwi-cli automation scenarios (we run Wokwi CI on GitHub Actions for our I²C tests)?
I'll report back on this issue when I've tried it.
Jérôme