wokwi/wokwi-features #226: Update LC_baseTools Please?

leftCoast, 2021-11-10:
There's a new version of LC_baseTools out there. ( Its in the IDE library manager ) Can I get it added please? There's some stuff in there I'd like to use on my stuff.

Thanks!

-jim

urish, 2021-11-10:
It should be automatically available if you add it to libraries.txt. You may have to specify a specific version number, e.g.

```
LC_baseTools@1.3
```

If this is the case (i.e. you get the latest version when you specify a version number, and a previous version when you don't) - let me know and I'll purge the cache.

Thanks!

leftCoast, 2021-11-10:
For some reason it wont "find" the 1.3 version of base tools.

https://wokwi.com/arduino/projects/314465668988666432

-jim

urish, 2021-11-10:
Interesting, seems like I have to update the library index for the change to be picked up.

I just updated it - can you try again? Note that you may have to change something in the code (add a comment or something) to bypass the cache.

leftCoast, 2021-11-10:
It worked once. I tried deleting the librarys.txt Now, no matter what I
do, it won't work again.

Same project.

-jim

urish, 2021-11-10:
This is a problem with the cache. Can you try changing the project (e.g. adding an empty comment), saving it, then refreshing the page 2-3 times?

leftCoast, 2021-11-11:
OK that works. I have to reload it a few times, then it takes. Looks like
its not an issue if your not changing library versions.

-jim

closed 2021-11-11 by leftCoast