wokwi/wokwi-features #228: Add rotary encoder with switch

elliotmade, 2021-11-15:
Please add a regular rotary encoder with a switch.  TMPCB  is using an encoder for the current course and this would be super.  Here is an example of the type of encoder that would meet our needs:
https://tech.alpsalpine.com/prod/e/html/encoder/incremental/ec11/ec11e09244bs.html

The actual part we're using in the course incorporates LEDs, so that would be a bonus, but not necessary unless you're feeling fancy:
PEL12T
https://www.bourns.com/docs/product-datasheets/PEL12T.pdf

Thanks!

urish, 2021-11-15:
Hi Elliot!

Have you checked out the https://docs.wokwi.com/parts/wokwi-ky-040 ? It's a rotary encoder with a switch.

Does it miss anything?

elliotmade, 2021-11-15:
In fact it's me that missed something - based on the pin names I assumed that was a serial output (in my defense, Mark also skipped over this encoder when we were briefly looking at the devices in the list too).

Oops!

closed 2021-11-15 by elliotmade