wokwi/wokwi-features #229: Local library list

leftCoast, 2021-11-16:
The issue:
I rely on a LOT of local library files for my stuff. These are libraries that are not ready to be released as "real" libraries for the Arduino IDE. I end up with so many file tabs, there is no longer room to edit my .ino file! Seriously! AND it messes up all the <> & "" stuff. Causing huge headaches going back and forth.

What I want :
locaLibrary.txt file to list my local libraries .h & .cpp than I want to include from my Arduino Library folder at my compile time. Files without tabs. Along with this i'd like some sort of refresh button/control to grab a fresh copy of these files, if available for the project.

urish, 2021-11-16:
I hear your pain.

I still have to think what the best way to solve this would be. The browser doesn't allow direct access to your computer's filesystem, so loading the libraries from a local directory might require you to reselect the directory every time, whenever a file changes. I'll have to look into this, but I think Safari doesn't even permit selecting a directory tree, just multiple files. Chrome allows directory selection (like we currently do for the SD card).

Anyway, I'll open it for voting. I'm sure will figure out the details as it goes up the list...

leftCoast, 2021-11-17:
Actually I was thinking about this should work this AM. This thing we are talking about needs to be not so much a file selection as a folder selection. Libraries come in folder form as a set of source code.

Could we set it up as a preference attached to your account? Something refreshable? Grr.. Not knowing the ins an outs of web development etc. I can't come up with helpful ideas. Its very frustrating.

urish, 2021-11-18:
Wow, that was fast. This request made it to the [top of the list](https://wokwi.com/features). 

I'll start researching that available options - which browsers support folder selection, which browsers can automatically reload the libs when you start the simulation (so you don't have to reselect the folder every single time). I suspect Chrome supports both, and Safari probably neither. But we'll know better once I test it.

Other than that, there are some changes needed in the build server to support this. 

As for the library list, I think we can still specify all libraries in `libraries.txt`. Wokwi would check if you have enabled a local libraries folder, and look there first. If it finds a matching library, it'll send it to the build server. Otherwise, it'll use the standard Arduino library manager. Thought on this?

urish, 2021-11-21:
Update: I have a sketch of a possible solution. It currently only works on Chrome:

1. Go to code editor, press "F1"
2. Select "Select local Arduino libraries directory"
3. Choose the directory with all your local Arduino libraries

Now when you build a project, the Wokwi will go over "libraries.txt" and match each entry against your local libraries. If it finds one, it'll upload it to the server and use it for the build. 

Can you try it and let me know if it works for you?

leftCoast, 2021-11-22:
Oh wait. I think I got it. I have to list every library ones from the IDE manager and everything I need form my own library folder. And it pulls them all in. Does it correctly use <> for the .h files from the library folder?

leftCoast, 2021-11-22:
Ok, I've set up a project. I've not put the display, SD drive or things like that on it. But, I have included my list of library folders I need to compile it.

The project : https://wokwi.com/arduino/projects/315994072755995201

It stops compiling on the first library that's NOT set up for the library manager of the Arduino IDE. (Most are not ready for release so they are jut piles of source code.)

urish, 2021-11-22:
> Can you explain it in a little more detail? What is the logic its going to use here?

Sorry for not explaining myself very well, it was like 2am when I wrote it :)

> Does it correctly use <> for the .h files from the library folder?

Yes. As if they were installed from the library manager.

> The project : https://wokwi.com/arduino/projects/315994072755995201

The project seems to be set up correctly. let's verify a few things:

1. Did you choose the "libraries" folder itself in "Select local Arduino libraries directory" - that should be a folder with subfolders for each of your local in-development libraries: LC_baseTools, LC_docTools, etc.
2. Does each of the library subfolders contain a "library.properties" file in it?

leftCoast, 2021-11-23:
Oh, I did NOT select the "libraries" folder. I didn't relize I could jut do that. That would be great if possible!

Most of the libraries do NOT have "library.properties" files in them. As I said, they are "in process" so they are just piles of .h & .cpp files in folders. When one is mature enough to release, I put the properties file in there and do the sub folder hierarchy thing.

leftCoast, 2021-11-23:
So I tried just listing "/Users/jimlee/Documents/Arduino/libraries" and didn't work at all. According to the java monitor it couldn't find it. That path is what I got from my linux terminal window by putting my working directory as that folder.

urish, 2021-11-23:
Is there any way you can share a quick recording of your screen so I see how you are doing this?

> Most of the libraries do NOT have "library.properties" files in them.
As far as I can tell, it the arduino-cli needs a library.properties file inside in order to consider the folder as a library folder. You can probably get away with just copying a library.properties file from an existing library.

leftCoast, 2021-11-27:
I still think it should be two.. no. **_three_** lists. 
1) The tabbed list you load manually for your sketch.
2) The libraies.txt list we now have for Arduino libraries.
3) The hidden list that is our Arduino library folder. We load this at the click of a "load local library" button.

This will keep everything straight and portable for sharing.

As for the SD drive? I'd love to handle it the same way, but I guess there's issues with that?

urish, 2021-11-28:
It's getting a bit complex. I'm thinking of an alternative solution: a "Library Manager" window where you can add standard Arduino libraries to your project, as well as load specific libraries from your computer. 

This will do two things:
1. Updates libraries.txt for you
2. Upload the local libraries you select from your computer to the server, and save them as part of the project. Users will be able to view them and download them to their computers (as ZIP files, one per library) through the "Library Manager".

Thoughts?

leftCoast, 2021-11-28:
Sounds good. Can we possibly do the SD card images as well using this?

urish, 2021-11-29:
> Sounds good. Can we possibly do the SD card images as well using this?

In a way. The infrastructure for uploading and storing the files will probably be shared, but SD card have somewhat different needs:

1. Support for binary file storage
2. They need to be downloaded to the browser before the simulation can starts. That's in contrast with libraries, which need to be downloaded to the compilation server (at least until the user requests to explicitly download a ZIP file with one of the local libraries).
3. Different storage requirements. SD card can have tens or even 100s megabytes of data (not sure we'll support this; downloading 100mb of data to your browser just to start the simulation sounds too much)
4. Probably a different user interface for managing the SD card content.

As you can see, as long as all the data is stored on only your PC, it's much easier to manage :-)

> Oh, and make sure the uploaded personal libraries do NOT need the #includes changed.

Sure thing

leftCoast, 2021-12-11:
"Different storage requirements. SD card can have tens or even 100s megabytes of data (not sure we'll support this; downloading 100mb of data to your browser just to start the simulation sounds too much)

Probably a different user interface for managing the SD card content."

How about we limit disk images? Say... 10Mb? I just looked at mine and its between 1..2 Mb I can see it could grow, but for now 10 would be fine. When/if people's needs get bigger, we can address that then.

And would we be able to write to the disk image? That would be an entire new can O' worms.

urish, 2021-12-12:
> And would we be able to write to the disk image? 

You mean, from within the simulation?

leftCoast, 2021-12-13:
Yeah, from within the simulation. You could leave messages.

urish, 2021-12-14:
I think writing would work. But then if you wanted to look at the files from outside the simulation, that's a whole other story :)

urish, 2021-12-25:
Feedback from @leftCoast: 

> While we're at it, can you default to having libraries.txt as one of the defaults? I really hate typing it in each time.

yepher, 2022-01-06:
I know this feature is called `...local directory....` but to me, that breaks the browser/file system, security model. Have you considered an alternative solution like using a "package manager" instead? Essentially an `npm` for Arduino. I did a quick google search and it looks like there is something called [`apm`](https://github.com/ksrichard/apm) that can be used with Arduino.

I think if done correctly it can preserve the great feature of the same code that works on Wokwi will also work locally. This is how I would see it works:

If there are no custom libraries everything works just like it does today.

If the user needs one or more custom libraries they needed to set those up on their local file system and add those to `libraries.txt`. In Wokwi (or locally) they would define the dependency definition file. The build server (or the user locally) runs the tool to pull dependencies and then does the compile.

This will not allow you to use files from your local file system. But as long as the dependency is available at some HTTP(s) address it would work. I would probably use GIT with various branches for a GIT repo so I could use it in developer or production versions of my library.

My concern for Wokwi operation team with either solution is the possibility of getting exposed to malicious code getting added to your build server.

leftCoast, 2022-01-06:
What is : "define the dependency definition file" ?

yepher, 2022-01-06:
@leftCoast That is a file where the person writing the code would describe what dependencies the code requires and where to get those dependencies. It is the same sort of thing you do with `package.json` for npm or `pom.xml` for Java/Maven or `Cartfile` for iOS/Carthage etc.

The exact format of that file would depend on what package manager was used. I've not used it but looks like `apm` is what some folks use for Arduino work.

leftCoast, 2022-01-06:
"It is the same sort of thing you do with..." And the rest just reads like gibberish to me.

I worry that this, whatever it is, may end up just a giant roadblock for people developing code to run on Wokwi. It certainly sounds like it would put a damper on what I'm doing. What's it supposed to be fixing/helping?

sutaburosu, 2022-01-06:
It's not that complicated, Jim. Chris seems to be describing the functionality we already have in [libraries.txt](https://docs.wokwi.com/guides/libraries) except without the search feature, which I understand is going to be added to Wokwi in the near future.

leftCoast, 2022-01-06:
@sutaburosu I'll trust you on that. Its all in an area I have to experience in. Thanks for chiming in.

urish, 2022-01-11:
There we go, club members can now upload their own, custom libraries:

[image]

And anyone can use the uploaded libraries in their projects (just copy the relevant entry in libraries.txt)

urish, 2022-01-14:
And now this also has documentation: https://docs.wokwi.com/guides/libraries#uploading-custom-libraries

closed 2022-01-14 by urish

leftCoast, 2022-01-15:
Congrats on a closed feature!