wokwi/wokwi-features #231: EEPROM persistence

Koepel, 2021-11-18:
Feature request: store the internal EEPROM somewhere to make its data persistent.

My request is a for the internal EEPROM of the AVR family boards. When returning to Wokwi next day, the previous written data should still be there.

**No extras**
There are many additions that I can think of (download the contents; have a number of binary files which can be choosen or uploaded; also store the simulated EEPROM of the SAMD and ESP processors; and so on), but that it too much over the top.

Writing to EEPROM has no clock-accurate delay. It is slower on a old ATmega8 then on a newer microcontroller. This might have consequences for interrupts. Since this is a very specific situation, I don't mind if the delay for writing data is skipped.

**Storage location**
It would be nice if the data was stored on the Wokwi server in the project and not on the user's computer. That allows to use an other computer and continue with the same data.

greenveg, 2021-12-08:
I too need better EEPROM functionality.
I mostly need start/stop functionality to behave as power off as I'm trying to write good power-loss recovery stuff.

Gave it 4 votes 👍

urish, 2021-12-12:
Thanks @greenveg! What project are you working on?

greenveg, 2021-12-13:
Using a [Controllino](https://www.controllino.com/) (Arduino framework PLC) for test rig at work. Need power-off recovery and such…

M-square, 2022-06-03:
(This comment is mostly a "bump" - but I also just used a few votes on it!)

urish, 2022-10-23:
Note: additional discussion in #399

ppvnf, 2023-05-10:
newbie here, for my school project IRL I am using `highScore = EEPROM.read(0);`, however in the emulator I have to use `EEPROM.get(0, highScore);` for some reason, otherwise I will get 255 as the high score and also the other way around if I use `highScore = EEPROM.read(0);` in the emulator

Koepel, 2023-05-10:
When you buy a new ATmega328P microcontroller (as used in the Arduino Uno), then the EEPROM is empty. A empty EEPROM means that all the data is 0xFF. Wokwi simulates a fresh new unused Arduino Uno and therefor the EEPROM is empty and is 0xFF.
The Arduino Uno you have real life is probably be used by others.

jumpjack, 2024-03-24:
How can I store EEPROM contents in browser local storage?

Koepel, 2024-03-24:
That is not implemented yet. The EEPROM contents is not stored in the Wokwi project as a file yet and it is not stored on your computer as local browser data.