wokwi/wokwi-features #233: Problems with generated VCD files with logic analyzer for gtkwave

bvandepo, 2021-11-22:
Dear Uri, you have made a great work with Wokwi, I'm using it with my students to teach them embedded software development.
In this context, I’m using the logic analyzer to capture signal (for timing measurements and checking).  First I had problems to open the generated VCD files with gtkwave for 2 reasons:

1) the date and time is generated on French Locale systems and appears as:
 $date Mon Nov 22 2021 17:41:45 GMT+0100 (heure normale d’Europe centrale) $end
instead of 
 $date Mon Nov 22 2021 17:41:45 GMT+0100$end

2) the carriage return and newline is sometime missing  at the end of the VCD file

I made a python script to correct these problems but also to completely automatize the opening of VCD files so my students have a gtkwave update each time they press the stop button in your amazing simulator.

I’ve stored my project at: https://github.com/bvandepo/wokwi2gtkwave
may be you could add a link to it in the documentation of the logic analyzer. I’ve tested it with Linux and Windows (and my 60 students did too)

Thanks and Congrats

urish, 2021-11-22:
Hi, nice to meet you!

I like the idea of automating opening of VCD files. I'll add a link to the docs.

Thanks for reporting the issues, I've uploaded a fix: now VCD files should always had with a newline, and the date is in the GMT timezone so it should be more friendly to gtkwave. Can you please check and report if now gtkwave can open the files on French Locale systems?

bvandepo, 2021-11-22:
Thank you for the reply.
I've just tested on an existing project and I get $date Mon Nov 22 2021 23:57:42 GMT+0100 (heure normale d’Europe centrale) $end and there is no new line at the end of the file even after rebooting.
Strangly, after I did a "save as" for the project, I captured a new VCD file and I obtained a correct $date Mon, 22 Nov 2021 23:04:38 GMT $end . I don't know if that comes from a caching problem... but I just have to "save as" all my projects. 
However,  I don't see the \n for the last line of the vcd file. I've observed strange behaviour of gtkwave when the autozoom feature is active and the \n is missing. gtkwave then sometime reports  a "Time backtracking" error. I've checked the date of every transitions in the file and they are in ascending order. Once I've added the \n, I never had this issue again. So you can add it if you want, or my script does it (for its particular usage with the autozoom activated at startup).

Thank you!

urish, 2021-11-23:
Thanks for checking!

Yes, this sounds like a caching issue. I can't tell why I get the \n at the end but you don't. Mind sending me one of the latest VCD files so I can have a look?

urish, 2021-12-22:
Closing due to inactivity. Feel free to reopen if still relevant.

closed 2021-12-22 by urish