wokwi/wokwi-features #235: Add ST boars support

SebCanet, 2021-12-07:
Thanks a lot for this great project!
To be a complete microcontroller simulator, I think you should add at leaste the 3 generic board from ST: nucleo32, nucleo64 & nucleo128.
Even if microcontroller is not exactly the same, at last it would be a starting point (and anybody could tweak to change few elements like memory size, etc) because pinout and features in same product range are similar.
Thanks.

urish, 2021-12-07:
Hi Seb, thanks for the feature request!

If you had to choose one board, which one would you go with?

SebCanet, 2021-12-07:
Even if ST is working on a board for education (https://www.st.com/en/evaluation-tools/p-nucleo-wb55.html), I think Nucleo64 is a good starting point as it's same pinout than Arduino Uno.

urish, 2021-12-07:
Thanks! Do you know to tell anything about the popularity of the different Nucleo64 variants ([source](https://www.st.com/resource/en/user_manual/um1724-stm32-nucleo64-boards-mb1136-stmicroelectronics.pdf))?

[image]

SebCanet, 2021-12-07:
I think the L476RG is a good candidate., or F401RE.
Arduino IDE can compile to it now, with STM32Duino project: [https://github.com/stm32duino](https://github.com/stm32duino)

urish, 2021-12-07:
How useful are the F0/L0 series are (F030R8, L010RB, etc.)?

SebCanet, 2021-12-07:
Not expert, I cannot answer...sorry.

urish, 2021-12-07:
That's okay. I'm asking because F0/L0 are based on Cortex-M0 core, which is already implemented as part of the [RP2040 emulator](https://github.com/wokwi/rp2040js). So they will probably be faster to implement (weeks instead of months of work).

SebCanet, 2021-12-07:
Good idea! That's a great starting point, many thanks!

urish, 2021-12-07:
Alright, I opened this feature request for [public voting](https://wokwi.com/features#feature-1073107644). Let's see how it goes!

closed 2022-07-28 by urish