wokwi/wokwi-features #236: Real LED response?

alto7, 2021-12-08:
Is it my imagiantion or do wokwi LEDs turn off about 100000 times slower than real LEDs?

If you have built in a fading effect turn-off, you have wonderfully emulated an incandescent lamp.

'If there is a fade effect, please add an attr "fade": "0" or however, I'll let you fight amongst yourself about whether that should really be the _default_ part.

'if there is no fade build in, then it is a curious artifact of the simulation, which makes using it for blinking LEDs less valuable, and me very curious about the psycho visual effect.

alto777

urish, 2021-12-08:
The thing is: real LEDs can update millions time a second, and usually have a constant brightness (at a given voltage level). So how do you create varying brightness on Arduino? with PWM.

Monitors can update 60 times per seconds (sometimes more), and can have different levels of brightness per pixel.

So, the workaround in the simulator is to look at the relative time the LED was on during the last monitor frame (1/60th of second), and use that as the brightness level for the LED.

This create subtle fading. However, it's not enough - since simulation can run slower than real time, so 1/60th of a real-time second is actually less in simulation time (at 10% that would be 1/600), which would create notable flickering with PWM (for example #62). The thing is - when the LED goes off, the simulator doesn't know the future. It doesn't know when the LED will turn on again, so we can only look at the past to determine the duty cycle of a PWM signal.

In short, this is a complicated matter - you can't accurately simulate the light of LEDs on a computer monitor in real time in a realistic way. That's why there's this trade-off: either have some fading, or flickering when you use PWM.

closed 2021-12-08 by urish

sutaburosu, 2021-12-08:
@alto7 you may be able to get closer to the visual effect you desire by reducing the LED's gamma correction factor to 1.  See the example sketch [in the gamma docs](https://docs.wokwi.com/parts/wokwi-led#gamma-correction).

alto777, 2021-12-09:
THX, that makes sense totally. Looks like some nasty potential for bad combinations of PWM or hand-programmed on/off and the rendering 60 Hertz.

I don't generally use PWM, except when it comes out of the box like with neopixels. Which I am guessing was an equal nightmare implement.

I'll give that a try @sutaburosu, thank you. For now I've just reduced the winking and blinking frequencies to yield enough truly off time and truly on time so they appear to be. Blinking and winking.

alto777

urish, 2021-12-09:
Interestingly, this causes an issue where blinking the built-in LED every 100ms causes it to be always-on: 

https://wokwi.com/arduino/projects/317535907128279616

I'll check if we can tweak the parameters to make it a bit better